esp-rs/esp-wifi-sys #337: Fix CAS requirements

TheButlah, 2023-11-06:
Currently, there are two dependencies that require CAS operations, which is undesirable when trying to avoid the atomic trap handler on esp32c3:

- `atomic_enum` - the generated code references CAS ops from the stdlib, even though we don't actually use any in this crate.
- `atomic-waker` - a transitive dependency, which uses CAS operations. Luckily they use `portable-atomics` which allows using emulated CAS ops. However, `esp-wifi`/`embedded_svc` don't actually enable the feature flags necessary to get those portable atomics to be used.

bugadani, 2023-11-06:
Since the crate author seems to have disappeared I made this instead: https://github.com/bugadani/portable_atomic_enum The plan was to use this crate instead of atomic_enum, but it doesn't compile if someone uses the atomic emulation on RISC-V. I guess a CAS feature would make sense for my fork, too, but we can't use it yet and have to resolve the ["emulation v portable-atomic"](https://github.com/esp-rs/riscv-atomic-emulation-trap/issues/11) situation in some way before that changes.

TheButlah, 2023-11-06:
Given that this functionality is not exposed to end-users of esp-wifi, I would expect that esp-rs can choose to use a fork or simply remove the dependency altogether without impacting any consumers of the library

DerFetzer, 2023-11-26:
@TheButlah I tried your workaround on one of my projects, but I still get a compilation error:

```
error[E0412]: cannot find type `AtomicUsize` in module `core::sync::atomic`
 --> /home/derfetzer/src/esp-wifi/esp-wifi/src/wifi/state.rs:7:1
  |
7 | #[atomic_enum]
  | ^^^^^^^^^^^^^^ not found in `core::sync::atomic`
  |
  = note: this error originates in the attribute macro `atomic_enum` (in Nightly builds, run with -Z macro-backtrace
 for more info)
help: consider importing one of these items
  |
1 + use atomic_polyfill::AtomicUsize;
  |
1 + use crate::wifi::AtomicUsize;
  |
1 + use portable_atomic::AtomicUsize;
  |

error[E0433]: failed to resolve: could not find `AtomicUsize` in `atomic`
 --> /home/derfetzer/src/esp-wifi/esp-wifi/src/wifi/state.rs:7:1
  |
7 | #[atomic_enum]
  | ^^^^^^^^^^^^^^ could not find `AtomicUsize` in `atomic`
  |
  = note: this error originates in the attribute macro `atomic_enum` (in Nightly builds, run with -Z macro-backtrace
 for more info)
help: consider importing one of these items
  |
1 + use atomic_polyfill::AtomicUsize;
  |
1 + use crate::wifi::AtomicUsize;
  |
1 + use portable_atomic::AtomicUsize;
  |

```

Do you have an idea what the problem could be?

TheButlah, 2023-11-26:
CAS is a specific subset of the full atomic capabilities, the compare_and_swap / compare_exchange functionality. Atomic load/stores can often be emulated entirely via volatile read/write operations, which require no special hardware support on single core devices. 

In my project, I can enable a subset of the full atomic features for the given target platform: https://github.com/SlimeVR/SlimeVR-Rust/blob/9eff429f4f01c8b7c607f3c3988de82729c753b3/firmware/.cargo/config.toml#L22C3-L30C49

This gives me access to std::sync::atomic::AtomicUsize without giving me access to the CAS operations.

PR #342 (2023-12-05): Update dependencies, port to portable-atomic
Blocked on the next esp-hal release. The PR keeps atomic emulation stuff enabled for ESP32-S2 because the old heapless doesn't know about xtensa and so it won't enable atomic-polyfill even with the `cas` feature enabled. Other than that, RISC-V targets should now be able to compile esp-wifi with the emulation trap disabled.

Closes #337
Closes #373
closed 2023-12-05 by MabezDev