esp-rs/esp-wifi-sys #341: Status of LTO

ProfFan, 2023-11-11:
I can see from #108 that LTO is not recommended for use. However from my benchmarks (not on ESP-wifi) my application has a 50% performance increase just from enabling LTO. What is the current roadblock for enabling LTO support? Thanks!

bugadani, 2023-11-12:
I've been using `lto = "fat"` for months now without issues. There have been significant improvements to the Xtensa compiler this year. The recommendation against it wasn't removed because I think actually validating this is pretty hard. More feedback regarding LTO should probably help in making a change here.

umgefahren, 2023-11-14:
So the issue is only affecting the xtensa toolchain? Are the RISC-V targets affected?

bugadani, 2023-11-14:
Oh I assumed so, but the linked issue seems to say it's both. I haven't worked with anything RISC-V yet so can't comment on that. Still, 9 months is a long time. If you have some time to experiment, this issue might be a good place to write feedback.

PR #363 (2023-11-28): Remove recommendation against LTO
Closes #341
closed 2023-11-28 by bjoernQ