esp-rs/esp-wifi-sys #351: esp32c6 ESP-NOW unable to set rate.

alexbohm, 2023-11-20:
I'm working on a project using esp now on the esp32c6 and I'm unable to set a rate with the `set_rate` function. 

I can reproduce this with the embassy_esp_now example by adding this line just before the loop.
```rust
esp_now.set_rate(esp_wifi::esp_now::WifiPhyRate::Rate2m).unwrap();
```

When I turn on the wifi logs, I get the message:
```
INFO - W (336) wifi:
INFO - This API is not supported in WIFI6 mode, please use esp_now_set_peer_rate_config() instead
INFO - 
```

It doesn't look like `esp_now_set_peer_rate_config()` is in the bindgen bindings, and according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/network/esp_now.html#config-esp-now-rate, that's what is supposed to be used to configure the rate for each added peer.

Looking at the header in esp-idf, this function was added fairly recently:
https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_now.h

alexbohm, 2023-12-07:
Thanks for the update!

I've been playing around with https://github.com/esp-rs/esp-wireless-drivers-3rdparty to see if I could update them myself while learning a bit more about how this is all integrated.

First I copied the new headers into esp-wifi-sys from esp-idf. This was pretty straight forward, I just matched the filenames.

Then, I played around with updating the libs using esp-wireless-drivers-3rdparty.
Most of the libs seemed to update fine except wpa_supplicant needed a couple more options disabled in sdkconfig to keep CONFIG_ESP_WIFI_MBEDTLS_CRYPTO (previously CONFIG_WPA_MBEDTLS_CRYPTO) disabled.

`patch/esp32c6/sdkconfig.defaults`:
```
CONFIG_ESP_WIFI_ENABLE_WPA3_SAE=n
CONFIG_ESP_WIFI_ENABLE_WPA3_OWE_STA=n
CONFIG_ESP_WIFI_MBEDTLS_TLS_CLIENT=n
CONFIG_ESP_WIFI_DPP_SUPPORT=n
CONFIG_ESP_WIFI_MBEDTLS_CRYPTO=n
CONFIG_COMPILER_OPTIMIZATION_SIZE=y
```

Those options let everything compile in esp-wireless-drivers-3rdparty, but in the rust compile it got hung up during linking with:
```
  = note: rust-lld: error: undefined symbol: sleep
          >>> referenced by os_xtensa.c:55 (/home/alex/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:55)
          >>>               os_xtensa.c.obj:(os_sleep) in archive /home/alex/code/esp-control/target/riscv32imac-unknown-none-elf/release/build/esp-wifi-sys-717724a08f01e6eb/out/libwpa_supplicant.a
          
          rust-lld: error: undefined symbol: usleep
          >>> referenced by os_xtensa.c:55 (/home/alex/esp/esp-idf/components/wpa_supplicant/port/os_xtensa.c:55)
          >>>               os_xtensa.c.obj:(os_sleep) in archive /home/alex/code/esp-control/target/riscv32imac-unknown-none-elf/release/build/esp-wifi-sys-717724a08f01e6eb/out/libwpa_supplicant.a
```
I'll have to poke a bit more at those functions later, I'm not sure if it should be trying os_xtensa.c anyways.

I did try using the old libwpa_supplicant.a, and the embassy esp now example did compile but it seemed to get stuck during the initialization process.

alexbohm, 2023-12-09:
Nice! I'm seeing the same here. esp-rs/esp-hal#1008 seems to stop the exceptions but something is still not quite right. It doesn't seem to receive anything from another esp running that same esp-now broadcast example.

I also tried the dhcp and embassy dhcp example and while they ran, it couldn't seem to connect to the network, giving a reason 201:
```
INFO - D (4236) wifi:
INFO - Send disconnect event, reason=201
INFO - 
```
I'm pretty sure I have my password correct, so according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/wifi.html, it's having trouble finding/scanning the access point.

bjoernQ, 2023-12-11:
At least progress - I think the newer drivers added some config parameters - are we sure we set values which make sense (e.g. compared to what esp-idf's defaults are)?

IIRC the esp-now example at least sends something which was shown on another device (running the example from `main`)

alexbohm, 2023-12-17:
I've been looking at some of the config options. tbh, I don't have enough experience with esp wifi stuff to know which ones to change. I compared them to what is in the esp idf, and it seems like they are set to appropriate defaults? I did switch one of the new options `rx_mgmt_buf_num` to the default of 5 but it didn't seem to change anything.

With one board running the old drivers and one board running the "updated" drivers, I ran the esp now example and the old drivers did receive the esp now messages from the updated drivers, so at least I know that the updated drivers are sending the messages.

bjoernQ, 2023-12-18:
Many of those settings are a mystery since they are often not well documented - I usually try to align them with the defaults in esp-idf in that case

Interesting it seems to be able to send the messages just fine but only fails to receive anything 😲 doesn't make me look forward to the next driver update after publishing 0.2

bjoernQ, 2024-01-11:
The drivers are updated now - unfortunately you will need to patch esp-hal for now (until the next esp-hal release)

alexbohm, 2024-01-21:
Thanks for getting the drivers updated! Out of curiosity, after updating did sending and receiving just work? Maybe I just had a bad build.

I've been exploring how to set the rates and it seems like some of the phymodes + rates may need protocol, bandwidth, and channel changes in order to use them. I created a draft pull request #418 with some hardcoded values for now.

I'm not sure what the best api is for setting the rate, whether it calls only `esp_now_set_peer_rate_config` or if it does the channel/bandwidth changes in that same function as well.

bjoernQ, 2024-01-22:
> Thanks for getting the drivers updated! Out of curiosity, after updating did sending and receiving just work? Maybe I just had a bad build.

Yes, it just worked for me