esp-rs/esp-wifi-sys #353: esp32c6 needs bigger heap than esp32s3

bugadani, 2023-11-22:
My S3 can enable wifi just fine with the default heap, but C6 immediately runs out of memory:

```
[INFO] - Initializing Wifi driver
[INFO] - esp-wifi configuration Config { rx_queue_size: 16, tx_queue_size: 16, static_rx_buf_num: 32, dynamic_rx_buf_num: 16, static_tx_buf_num: 12, dynamic_tx_buf_num: 16, ampdu_rx_enable: 1, ampdu_tx_enable: 1, amsdu_tx_enable: 0, rx_ba_win: 8, max_burst_size: 0, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 200, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
[WARN] - Unable to allocate 2216 bytes
[WARN] - esp_wifi_init_internal(&G_CONFIG) returned an error: 257
[ERROR] - panicked at 'unwrap failed: esp_wifi ::
[ERROR] - initialize(EspWifiInitFor :: Wifi, resources.timer, resources.rng,
[ERROR] - resources.rcc, clocks,)'
[ERROR] - error: `WifiError(InternalError(EspErrNoMem))`
```

Increasing the heap size to `98304` immediately resolves this problem. My guess is that the various buffers might take up more space, maybe due to a different feature set, but might be good to dig into it a bit more.

MabezDev, 2024-02-20:
The heap is now configurable via the config so I think we can close this now.

bugadani, 2024-02-22:
The heap size has been configurable some time before I opened this issue. I understand it's not exactly actionable, but the 1.5x memory requirement is still something that at least should be investigated, or at least acknowledge the problem. Because reserving a bit more than a quarter of available RAM may be a problem.