esp-rs/esp-wifi-sys #355: Standalone "Embassy DHCP" example dependecies

windoze, 2023-11-26:
I created a new project with the code directly copied from [the embassy DHCP example](https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/examples/embassy_dhcp.rs) with only one line of change: `use esp32s3_hal as hal;`, but I failed to get it compiled, the error message is:

```
error[E0433]: failed to resolve: could not find `executor` in `embassy`
  --> src/main.rs:23:1
   |
23 | #[main]
   | ^^^^^^^ could not find `executor` in `embassy`
   |
   = note: this error originates in the attribute macro `main` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing one of these items
   |
5  + use embassy_executor::Executor;
   |
5  + use embassy_executor::raw::Executor;
   |

error[E0282]: type annotations needed
  --> src/main.rs:23:1
   |
23 | #[main]
   | ^^^^^^^ cannot infer type
   |
   = note: this error originates in the attribute macro `main` (in Nightly builds, run with -Z macro-backtrace for more info)

Some errors have detailed explanations: E0282, E0433.
For more information about an error, try `rustc --explain E0282`.
error: could not compile `esp-wifi-test` (bin "esp-wifi-test") due to 2 previous errors
```

below is the content of `Cargo.toml`

```
[package]
name = "esp-wifi-test"
version = "0.1.0"
authors = ["Chen Xu <windoze@0d0a.com>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[profile.dev.package.esp-wifi]
opt-level = 3

[profile.dev]
lto = "off"
[profile.release]
lto = "off"

[dependencies]
# esp-wifi requirements
esp32s3-hal = { version = "0.13.0", features = ["embassy", "embassy-time-timg0", "async", "log"]}
esp-backtrace = { version = "0.9.0", features = ["esp32s3", "panic-handler", "exception-handler", "print-uart"] }
esp-println = { version = "0.7.0", features = ["esp32s3","log"] }
esp-wifi  = { git = "https://github.com/esp-rs/esp-wifi/", rev = "0db7a70", features = ["esp32s3", "embedded-svc", "wifi-logs", "wifi", "async", "embassy-net", "phy-enable-usb"] }

defmt = "=0.3.5"
smoltcp = { version = "0.10.0", default-features=false, features = ["medium-ethernet", "socket-raw", "proto-ipv4", "socket-tcp", "socket-udp", "socket-icmp", "proto-igmp", "proto-dns", "socket-dns", "proto-dhcpv4", "socket-dhcpv4"] }
critical-section = "1.1.1"
log = "0.4.18"
embedded-svc = { version = "0.26.1", default-features = false, features = [] }
enumset = { version = "1", default-features = false }
linked_list_allocator = { version = "0.10.5", default-features = false, features = ["const_mut_refs"] }
embedded-io = "0.6.1"
embedded-io-async = { version = "0.6.0" }
fugit = "0.3.7"
heapless = { version = "0.7.16", default-features = false }
esp-wifi-sys = { git = "https://github.com/esp-rs/esp-wifi/", rev = "0db7a70", features = ["esp32s3"] }
embassy-sync = { version = "0.4.0" }
embassy-futures = { version = "0.1.0" }
toml-cfg = "0.1.3"
libm = "0.2.7"
cfg-if = "1.0.0"
static_cell = { version = "=1.2", features = ["nightly"] }

embassy-net = { version = "0.2.1", features = ["nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }
embassy-executor = { version = "=0.3.2", package = "embassy-executor", features = ["nightly", "executor-thread", "integrated-timers", "arch-xtensa"] } # temporarily pin because we aren't ready for portable-atomic yet
embassy-time = { version = "0.1.3", features = ["nightly"] }
futures-util = { version = "0.3.28", default-features = false }
atomic-polyfill = "1.0.2"
atomic_enum = "0.2.0" # uses core atomic types which should be fine as long as only load and store is used on them
```

I really have no idea why it cannot be compiled, any help is much appreciated.

bugadani, 2023-11-26:
You need to enable `embassy-executor-thread` on `esp32-hal` (and to remove `executor-thread` from embassy). esp-hal provides its own executors and main macro that you should prefer.

windoze, 2023-11-26:
Works like a charm. Thanks @bugadani.

Before closing this issue, may I make a suggestion to put some more detailed docs about how to use all the features? I was lost in the maze of configurations, and I believe that many others may have the same experience.
A `cargo-generate` template could be very helpful as well.

windoze, 2023-11-27:
Thanks for pointing out the right direction.

I'm looking at some improvements to the example and the template.

closed 2023-11-27 by windoze