esp-rs/esp-wifi-sys #357: build.rs panic with non-xtensa downstream version

xbjfk, 2023-11-27:
Hello, I am trying to use this crate on Gentoo, and the hack in `build.rs` is broken:
```
  --- stdout
  cargo:rustc-cfg=esp32c3

  --- stderr
  thread 'main' panicked at /home/rgn/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/7632e74/esp-wifi/build.rs:103:60:
  called `Result::unwrap()` on an `Err` value: ParseIntError { kind: InvalidDigit }
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
I suspect the reason for this is the gentoo version string looks like this:
```
rustc 1.73.0-nightly (cc66ad468 2023-10-03) (gentoo)
```

bugadani, 2023-11-27:
Hi can you please try with https://github.com/esp-rs/esp-wifi/pull/358 ?

closed 2023-11-27 by bjoernQ

xbjfk, 2023-12-03:
Yes, this works! Thank you very much for your quick response and fix!