esp-rs/esp-wifi-sys #374: Unable to compile for esp32 target.

Aib0t, 2023-12-05:
Hi.

Stumble upon an issue, which seems to be affecting only esp32 chip and related to ld linking.

Problem occurs when there is `initialize` in the code:

```rust

#[main]
async fn main(spawner: Spawner) -> ! {
    #[cfg(feature = "log")]
    esp_println::logger::init_logger(log::LevelFilter::Info);

    let peripherals = Peripherals::take();

    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();

    let server_address: Ipv4Address = HOST_IP.parse().expect("Invalid HOST_IP address");

    #[cfg(target_arch = "xtensa")]
    let timer = hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
    #[cfg(target_arch = "riscv32")]
    let timer = hal::systimer::SystemTimer::new(peripherals.SYSTIMER).alarm0;

    let init = initialize(
        EspWifiInitFor::Wifi,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();
    ```
    
    It produce errors like:
    
    ```
    ...
    home/aibot/Documents/personal/esp-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-813a0f22774e526d/out/libnet80211.a(ieee80211_nvs.o):(.text.wifi_nvs_set+0x1c): undefined reference to `strncpy'
    ...
    ```
    
 Full log file: [xtensa_esp32_log.txt](https://github.com/esp-rs/esp-wifi/files/13560193/xtensa_esp32_log.txt)

What I've tried so far:
- Switching from Windows to Linux produce exactly the same issue
- Downgrading build toold via `espup` doesn't help (Tried versions from 1.74.0.1 to 1.69.0.0)

I failed to find any info regarding the issue or something similar in this or `esp-hal` repo.

Aib0t, 2023-12-05:
That's what I've been doing so far. All of them fails similarly. I can provide my `Cargo.toml` if needed.

MabezDev, 2023-12-05:
I mean can you run them from within this repo, instead of copying them to your project?

Aib0t, 2023-12-05:
Building via 
```
SSID=TEST PASSWORD=TEST cargo esp32 --release --example=dhcp --features=wifi
```
does work but it doesn't build `esp-test`, which fails in my case.

```
   Compiling compiler_builtins v0.1.101
   Compiling core v0.0.0 (/home/aibot/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.70
   Compiling unicode-ident v1.0.12
   Compiling syn v1.0.109
   Compiling serde v1.0.193
   Compiling portable-atomic v1.5.1
   Compiling strsim v0.10.0
   Compiling fnv v1.0.7
   Compiling ident_case v1.0.1
   Compiling heck v0.4.1
   Compiling semver v1.0.20
   Compiling autocfg v1.1.0
   Compiling rustversion v1.0.14
   Compiling anyhow v1.0.75
   Compiling futures-core v0.3.29
   Compiling memchr v2.6.4
   Compiling futures-task v0.3.29
   Compiling version_check v0.9.4
   Compiling futures-util v0.3.29
   Compiling embassy-time v0.1.5
   Compiling regex-syntax v0.8.2
   Compiling xtensa-lx v0.8.0
   Compiling hashbrown v0.14.3
   Compiling strum v0.24.1
   Compiling equivalent v1.0.1
   Compiling toml_datetime v0.6.5
   Compiling winnow v0.5.24
   Compiling heapless v0.8.0
   Compiling paste v1.0.14
   Compiling embassy-executor v0.3.3
   Compiling esp32 v0.28.0
   Compiling futures-channel v0.3.29
   Compiling smoltcp v0.10.0
   Compiling esp-println v0.7.1
   Compiling embedded-io-async v0.6.1
   Compiling esp32-hal v0.16.0 (https://github.com/esp-rs/esp-hal.git?rev=22f14fd#22f14fd3)
   Compiling libm v0.2.8
   Compiling uuid v1.6.1
   Compiling esp-backtrace v0.9.0
   Compiling proc-macro-error-attr v1.0.4
   Compiling proc-macro-error v1.0.4
   Compiling lock_api v0.4.11
   Compiling num-traits v0.2.17
   Compiling rustc_version v0.4.0
   Compiling aho-corasick v1.1.2
   Compiling esp-wifi-sys v0.1.0 (/home/aibot/Documents/personal/esp-wifi/esp-wifi-sys)
   Compiling indexmap v2.1.0
   Compiling quote v1.0.33
   Compiling litrs v0.4.1
   Compiling litrs v0.2.3
   Compiling heapless v0.7.17
   Compiling syn v2.0.39
   Compiling toml_edit v0.20.7
   Compiling regex-automata v0.4.3
   Compiling proc-macro-crate v2.0.0
   Compiling darling_core v0.20.3
   Compiling regex v1.10.2
   Compiling serde_derive v1.0.193
   Compiling strum_macros v0.25.3
   Compiling portable_atomic_enum_macros v0.2.0
   Compiling num-derive v0.4.1
   Compiling strum_macros v0.24.3
   Compiling enum-as-inner v0.4.0
   Compiling bleps-macros v0.1.0 (https://github.com/bjoernQ/bleps?rev=0db8fcb#0db8fcb7)
   Compiling darling_macro v0.20.3
   Compiling darling v0.20.3
   Compiling enumset_derive v0.8.1
   Compiling xtensa-lx-rt-proc-macros v0.2.1
   Compiling embassy-macros v0.2.1
   Compiling esp-hal-procmacros v0.7.0 (https://github.com/esp-rs/esp-hal.git?rev=22f14fd#22f14fd3)
   Compiling core-isa-parser v0.2.0
   Compiling rustc-std-workspace-core v1.99.0 (/home/aibot/.rustup/toolchains/esp/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling byteorder v1.5.0
   Compiling stable_deref_trait v1.2.0
   Compiling critical-section v1.1.2
   Compiling futures-sink v0.3.29
   Compiling pin-utils v0.1.0
   Compiling pin-project-lite v0.2.13
   Compiling scopeguard v1.2.0
   Compiling nb v1.1.0
   Compiling bare-metal v1.0.0
   Compiling log v0.4.20
   Compiling cfg-if v1.0.0
   Compiling embedded-hal v1.0.0-rc.1
   Compiling void v1.0.2
   Compiling mutex-trait v0.2.0
   Compiling embedded-io v0.6.1
   Compiling vcell v0.1.3
   Compiling enumset v1.1.3
   Compiling nb v0.1.3
   Compiling r0 v1.0.0
   Compiling gcd v2.3.0
   Compiling embedded-dma v0.2.0
   Compiling bitflags v2.4.1
   Compiling bitfield v0.14.0
   Compiling embedded-hal v0.2.7
   Compiling strum v0.25.0
   Compiling embedded-hal-async v1.0.0-rc.1
   Compiling bitflags v1.3.2
   Compiling fugit v0.3.7
   Compiling hash32 v0.2.1
   Compiling hash32 v0.3.1
   Compiling futures-io v0.3.29
   Compiling no-std-net v0.5.0
   Compiling managed v0.8.0
   Compiling linked_list_allocator v0.10.5
   Compiling spin v0.9.8
   Compiling atomic-waker v1.1.2
   Compiling portable_atomic_enum v0.3.0
   Compiling static_cell v2.0.0
   Compiling basic-toml v0.1.7
   Compiling minijinja v1.0.10
   Compiling toml v0.5.11
   Compiling esp-hal-common v0.13.1 (https://github.com/esp-rs/esp-hal.git?rev=22f14fd#22f14fd3)
   Compiling toml-cfg v0.1.3
   Compiling futures v0.3.29
   Compiling bleps v0.1.0 (https://github.com/bjoernQ/bleps?rev=0db8fcb#0db8fcb7)
   Compiling embedded-svc v0.26.4
   Compiling esp-wifi v0.1.1 (/home/aibot/Documents/personal/esp-wifi/esp-wifi)
   Compiling xtensa-lx-rt v0.16.0
    Finished release [optimized + debuginfo] target(s) in 17.01s
    ```
    
    My project was made via `cargo generate esp-rs/esp-template`, which might explain why there is a `esp-test` crate.

MabezDev, 2023-12-05:
It's likely the template needs updating, which will most likely happen with the next hal releases.

With that said, does adding this flag to your rustflags help: https://github.com/esp-rs/esp-wifi/blob/d46dfc4a7de5eb30ebc3b644d1abeaa52bc0b64a/esp-wifi/.cargo/config.toml#L25?

Aib0t, 2023-12-05:
Yes, it does! Thank you very much for the help with this unrelated issue.

closed 2023-12-05 by Aib0t

MabezDev, 2023-12-05:
Glad it's working now! :)