esp-rs/esp-wifi-sys #375: 20 timers may not be enough for C6

bugadani, 2023-12-05:
I'm getting this error:

`[WARN] - Failed to set timer function 40824b98`

The set of features I have enabled on esp-wifi are as follows, so I don't have BLE and coex enabled (meaning the library should be limited to 20 timers)

```
features = [
    "phy-enable-usb",
    "ps-max-modem",
    "wifi",
    "async",
    "embassy-net",
    "tcp",
    "dns",
    "ipv4",
    "dhcpv4"
]
```

bugadani, 2023-12-05:
This issue causes my C6 to not receive an IP via DHCP. Increasing the capacity [here](pub(crate) static mut TIMERS: heapless::Vec<Timer, 20> = heapless::Vec::new();) to 21 fixes the issue

bugadani, 2023-12-06:
I think we are at the point where we could instead start to heap-allocate these things.

bjoernQ, 2023-12-06:
Also an option