esp-rs/esp-wifi-sys #378: Cannot integrate esp-wifi into std project based on esp-idf-template

CliffHan, 2023-12-07:
To re-produce:

1. follow the `The Rust on ESP book`, create a project via: `cargo generate esp-rs/esp-idf-template cargo`, confirm it could work.
2. follow the README of esp-wifi, add dependency and compile args.

Here's the error log I got:
```
   Compiling esp-idf-hal v0.42.5
error[E0053]: method `acquire` has an incompatible type for trait
   --> C:\Users\Cliff\.cargo\registry\src\rsproxy.cn-8f6827c7555bfaf8\esp-idf-hal-0.42.5\src\task.rs:749:28
    |
749 |         unsafe fn acquire() {
    |                            ^ expected `u8`, found `()`
    |
    = note: expected signature `unsafe fn() -> u8`
               found signature `unsafe fn()`
help: change the output type to match the trait
    |
749 |         unsafe fn acquire() -> u8 {
    |                             +++++

error[E0053]: method `release` has an incompatible type for trait
   --> C:\Users\Cliff\.cargo\registry\src\rsproxy.cn-8f6827c7555bfaf8\esp-idf-hal-0.42.5\src\task.rs:753:35
    |
753 |         unsafe fn release(_token: ()) {
    |                                   ^^
    |                                   |
    |                                   expected `u8`, found `()`
    |                                   help: change the parameter type to match the trait: `u8`
    |
    = note: expected signature `unsafe fn(u8)`
               found signature `unsafe fn(())`

For more information about this error, try `rustc --explain E0053`.
error: could not compile `esp-idf-hal` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
```

I know the problem may comes with "std" feature. But I don't know how to solve it.

bjoernQ, 2023-12-07:
esp-wifi is to be used with the bare-metal hal (esp-hal). For esp-idf-hal you don't even need it

See https://esp-rs.github.io/std-training/03_3_2_http_client.html

closed 2023-12-07 by bjoernQ

CliffHan, 2023-12-07:
> esp-wifi is to be used with the bare-metal hal (esp-hal). For esp-idf-hal you don't even need it
> 
> See https://esp-rs.github.io/std-training/03_3_2_http_client.html

Looks like that's just an http client example, which used pre-defined wifi configuration.
Seems that example doesn't allow me to control wifi connection at runtime.