esp-rs/esp-wifi-sys #384: compile esp32c3 with embassy error

zrz4066, 2023-12-12:
I compile  esp32c3 with embassy ,there are duplicate symbol

= note: rust-lld: error: duplicate symbol: SYSTIMER_TARGET0
          >>> defined at riscv.rs:60 (src\timer\riscv.rs:60)
          >>>            esp_wifi-0dd8fd2238cf3dee.esp_wifi.40656898d412960c-cgu.09.rcgu.o:(SYSTIMER_TARGET0) in archive E:\Rust\GUI\esp32c3-slint\esp32c3-embassy\target\riscv32imc-unknown-none-elf\debug\deps\libesp_wifi-0dd8fd2238cf3dee.rlib
          >>> defined at time_driver_systimer.rs:75 (src\embassy\time_driver_systimer.rs:75)
          >>>            esp_hal_common-0b84c1864b31ad9e.esp_hal_common.4ca96403f295107b-cgu.14.rcgu.o:(.text.SYSTIMER_TARGET0+0x0) in archive E:\Rust\GUI\esp32c3-slint\esp32c3-embassy\target\riscv32imc-unknown-none-elf\debug\deps\libesp_hal_common-0b84c1864b31ad9e.rlib

src\timer\riscv.rs
[image]

src\embassy\time_driver_systimer.rs
[image]

E:\Rust\GUI\esp32c3-slint\esp32c3-embassy\target\riscv32imc-unknown-none-elf\debug\deps\libesp_wifi-0dd8fd2238cf3dee.rlib
          >>> defined at time_driver_systimer.rs:75 (src\embassy\time_driver_systimer.rs:75)
          >>>            esp_hal_common-0b84c1864b31ad9e.esp_hal_common.4ca96403f295107b-cgu.14.rcgu.o:(.text.SYSTIMER_TARGET0+0x0) in archive E:\Rust\GUI\esp32c3-slint\esp32c3-embassy\target\riscv32imc-unknown-none-elf\debug\deps\libesp_hal_common-0b84c1864b31ad9e.rlib

[image]

bjoernQ, 2023-12-12:
You are probably using the `embassy-time-systick` feature for esp-hal. That feature isn't compatible with esp-wifi - you can use `embassy-time-timg0` instead

closed 2023-12-13 by zrz4066

CM-IV, 2024-01-27:
esp-wifi with `esp32c3` feature expects `embassy-time-systick`, no?

I can use `embassy-time-timg0` to initialize wifi with an ESP32 device, but it seems like `embassy-time-systick` is needed for the RISC-V ESP32C3.

[image]