esp-rs/esp-wifi-sys #389: Can't build `cargo generate esp-rs/esp-idf-template cargo` + `cargo add esp-wifi -F esp32s3`

coder0xff, 2023-12-17:
I first ran `cargo generate esp-rs/esp-idf-template cargo`. I chose the esp32s3 target and the default options. I was able to `cargo build` fine. Seeing that the Bluetooth support wasn't included, I then ran `cargo add esp-wifi -F esp32s3`. When I run `cargo build` I see the following.
```
...
   Compiling esp-wifi-sys v0.1.0
error[E0432]: unresolved import `core::sync::atomic::AtomicU64`
   --> /home/coder0xff/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-polyfill-1.0.3/src/polyfill.rs:6:17
    |
3   | / macro_rules! atomic_int {
4   | |     ($int_type:ident,$atomic_type:ident, $cfg:ident) => {
5   | |         #[cfg(not($cfg))]
6   | |         pub use core::sync::atomic::$atomic_type;
    | |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `AtomicU64` in `sync::atomic`
...   |
163 | |     };
164 | | }
    | |_- in this expansion of `atomic_int!`
...
169 |   atomic_int!(u64, AtomicU64, polyfill_u64);
    |   -----------------------------------------
    |   |                |
    |   |                help: a similar name exists in the module: `AtomicU32`
    |   in this macro invocation

error[E0432]: unresolved import `core::sync::atomic::AtomicI64`
   --> /home/coder0xff/.cargo/registry/src/index.crates.io-6f17d22bba15001f/atomic-polyfill-1.0.3/src/polyfill.rs:6:17
    |
3   | / macro_rules! atomic_int {
4   | |     ($int_type:ident,$atomic_type:ident, $cfg:ident) => {
5   | |         #[cfg(not($cfg))]
6   | |         pub use core::sync::atomic::$atomic_type;
    | |                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `AtomicI64` in `sync::atomic`
...   |
163 | |     };
164 | | }
    | |_- in this expansion of `atomic_int!`
...
174 |   atomic_int!(i64, AtomicI64, polyfill_i64);
    |   -----------------------------------------
    |   |                |
    |   |                help: a similar name exists in the module: `AtomicI32`
    |   in this macro invocation

For more information about this error, try `rustc --explain E0432`.
error: could not compile `atomic-polyfill` (lib) due to 2 previous errors
```

coder0xff, 2023-12-17:
Looks like the atomic-polyfill issue is addressed by d46dfc4a7de5eb30ebc3b644d1abeaa52bc0b64a with #388 release 0.2.0. After, the question as stated in the title is still TBD. It's not easy to find the "right way" to accomplish this.

MabezDev, 2023-12-18:
esp-wifi is not meant to be used with esp-idf crates.

closed 2023-12-18 by MabezDev