esp-rs/esp-wifi-sys #395: Wi-Fi interferes with RMT peripheral

darkwater, 2024-01-01:
I'm using an esp32c3 to drive 100 WS2812 LEDs. A simple rainbow animation at 60 Hz looks fine until I initialize esp-wifi, when the colors start to flicker once or twice per second or so.

I suspect there are interrupts happening with a higher priority than the RMT peripheral, causing some buffer swap to be delayed. Reading some posts from non-Rust users gives me the same idea. The RMT transmit call is blocking so I don't think any async tasks are stalling.

bjoernQ, 2024-01-02:
My guess is that the problem is caused by the scheduling of the necessary tasks which need to run to make WiFi work

By default, the tick-rate (see https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/docs/tuning.md and https://github.com/esp-rs/esp-wifi/blob/a69545dc4c053b9cc9a7398df9250d16834526b5/esp-wifi/src/lib.rs#L92) is 100 Hz so it might take a long time until the main task gets a chance to run again. You can try another tick-rate (configured in `cfg.toml`) and see if that is enough to get it working

darkwater, 2024-01-07:
It looks like it helps a little bit, although it's hard to tell for sure. Even at 1 Hz it still flickers. I've recorded two videos to give a better idea of what it looks like.

https://github.com/esp-rs/esp-wifi/assets/1228306/1936dd3c-0fcf-42d5-8a43-b250e41de3ec

https://github.com/esp-rs/esp-wifi/assets/1228306/ba4a5be9-ad23-47f4-8522-94ce25a40abe

MabezDev, 2024-01-08:
RMT is very timing-sensitive, as @bjoernQ is suggesting. In esp-idf there are configuration options to store the RMT code in RAM, so that cache misses don't interrupt the transmission (we don't have an equivalent in Rust yet :(). Increasing the priority might also help here, but as @bjoernQ mentioned if there are long critical_sections there will still be flickering.

MabezDev, 2024-01-09:
Is your code open source? It might be easier to help if we can see the code. Other than that I can only suggest general advice.

bjoernQ, 2024-01-15:
A simple fix might be to run https://github.com/darkwater/home-leds/blob/48f39c9e8b55ffb0704709e077bdb8b5e3c09aa0/firmware/src/ws2812_driver.rs#L79 in a critical section - it should be fast enough to not interfere with esp-wifi

darkwater, 2024-01-21:
Thanks, that does solve the flickering. I'll open an issue on the hal repo because that logic should probably be done in an interrupt handler, or at least documented to be susceptible to interference.

closed 2024-01-21 by darkwater