esp-rs/esp-wifi-sys #399: Tuning cfg.toml doesn't work as expected

herve-ves, 2024-01-02:
Hi,
I try to play ESP32C6 with no-std esp-wifi and coex profile. It paniced when initialized the BLE because of OOM, and I followed 'tuning.md' to create a cfg.toml and set 'heap_size' to 200K. After that, it still paniced by OOM error. The logging is attached below:
```
Chip type:         esp32c6 (revision v0.0)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi 6, BT 5
MAC address:       40:4c:ca:41:29:90
App/part. size:    1,387,520/4,128,768 bytes, 33.61%
[00:00:00] [========================================]      13/13      0x0                                                                                                                                                 
[00:00:00] [========================================]       1/1       0x8000                                                                                                                                              
[00:00:18] [========================================]     631/631     0x10000                                                                                                                                             [2024-01-02T15:52:12Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
0x4086c410 - _stack_end
    at ??:??
load:0x4086e610,len:0x2d68
0x4086e610 - _stack_start
    at ??:??
load:0x40875720,len:0x1800
0x40875720 - _stack_start
    at ??:??
SHA-256 comparison failed:
Calculated: aff89878a96cbff57c66c38aa6d1a422b7785040b5efea582babef6c357427c2
Expected: 0af544a033ab3492852b8232c904c578d5f07c9d4a423a64473f060db374ab32
Attempting to boot anyway...
entry 0x4086c410
0x4086c410 - _stack_end
    at ??:??
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (47) boot.esp32c6: SPI Speed      : 40MHz
I (52) boot.esp32c6: SPI Mode       : DIO
I (57) boot.esp32c6: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=11fcfch (1178876) map
I (352) esp_image: segment 1: paddr=0012fd24 vaddr=40800000 size=07ca4h ( 31908) load
I (360) esp_image: segment 2: paddr=001379d0 vaddr=421279d0 size=27cc4h (163012) map
I (394) esp_image: segment 3: paddr=0015f69c vaddr=40807ca4 size=013f8h (  5112) load
I (396) esp_image: segment 4: paddr=00160a9c vaddr=408090a0 size=01e68h (  7784) load
I (402) esp_image: segment 5: paddr=0016290c vaddr=40828980 size=002c4h (   708) load
I (411) boot: Loaded app from partition at offset 0x10000
I (415) boot: Disabling RNG early entropy source...
Hello world!
 
 
!! A panic occured in '/home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/ble/npl.rs', at line 1070, column 13
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        ble_controller_init returned 257,
    ),
    location: Location {
        file: "/home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/ble/npl.rs",
[... 117 lines cut ...]
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=11fb20h (1178400) map
I (352) esp_image: segment 1: paddr=0012fb48 vaddr=40800000 size=07cb0h ( 31920) load
I (360) esp_image: segment 2: paddr=00137800 vaddr=42127800 size=28a04h (166404) map
I (394) esp_image: segment 3: paddr=0016020c vaddr=40807cb0 size=013ech (  5100) load
I (396) esp_image: segment 4: paddr=00161600 vaddr=408090a0 size=01e68h (  7784) load
I (402) esp_image: segment 5: paddr=00163470 vaddr=4083fa80 size=002c4h (   708) load
I (411) boot: Loaded app from partition at offset 0x10000
I (415) boot: Disabling RNG early entropy source...
Hello world!
Start connection task. Device capabilities: Ok(EnumSet(Client))
Starting wifi
Wifi started!
About to connect...
Wifi connected!
Connector created
Ok(CommandComplete { num_packets: 1, opcode: 3075, data: [0] })
Ok(CommandComplete { num_packets: 1, opcode: 8198, data: [0] })
Ok(CommandComplete { num_packets: 1, opcode: 8200, data: [0] })
Ok(CommandComplete { num_packets: 1, opcode: 8202, data: [0] })
started advertising
```

What is the difference between git revision patch and local repository patch?

bjoernQ, 2024-01-03:
Only thing I could think of is that the cfg wasn't picked up in the first try.

With info-level debug esp-wifi should also show what the current config looks like:
```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
```

herve-ves, 2024-01-03:
Yes, the cfg wasn't picked up in my first try. The logging  is attached below:
```
I (42) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (42) boot: compile time Jun  7 2023 08:02:08
I (43) boot: chip revision: v0.0
I (47) boot.esp32c6: SPI Speed      : 40MHz
I (52) boot.esp32c6: SPI Mode       : DIO
I (57) boot.esp32c6: SPI Flash Size : 4MB
I (61) boot: Enabling RNG early entropy source...
I (67) boot: Partition Table:
I (70) boot: ## Label            Usage          Type ST Offset   Length
I (78) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (85) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  2 factory          factory app      00 00 00010000 003f0000
I (100) boot: End of partition table
I (104) esp_image: segment 0: paddr=00010020 vaddr=42000020 size=122f10h (1191696) map
I (355) esp_image: segment 1: paddr=00132f38 vaddr=40800000 size=07cb0h ( 31920) load
I (363) esp_image: segment 2: paddr=0013abf0 vaddr=4212abf0 size=287d4h (165844) map
I (398) esp_image: segment 3: paddr=001633cc vaddr=40807cb0 size=013ech (  5100) load
I (400) esp_image: segment 4: paddr=001647c0 vaddr=408090a0 size=01e68h (  7784) load
I (405) esp_image: segment 5: paddr=00166630 vaddr=40828980 size=002c4h (   708) load
I (414) boot: Loaded app from partition at offset 0x10000
I (418) boot: Disabling RNG early entropy source...
Hello world!
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 50, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
WARN - Unable to allocate 280 bytes
 
 
!! A panic occured in '/home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/ble/npl.rs', at line 1070, column 13
 
PanicInfo {
    payload: Any { .. },
    message: Some(
        ble_controller_init returned 257,
    ),
    location: Location {
        file: "/home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/ble/npl.rs",
        line: 1070,
        col: 13,
    },
    can_unwind: true,
    force_no_backtrace: false,
}
 
Backtrace:
 
0x420510d2
0x420510d2 - esp_wifi::ble::npl::ble_init
    at /home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/ble/npl.rs:1070
0x42041050
0x42041050 - esp_wifi::initialize
    at /home/herve/.cargo/git/checkouts/esp-wifi-836f3b2af57fa847/a69545d/esp-wifi/src/lib.rs:301
```

I performed command 'cargo clean' to my workspace, then recompiled and this issue didn't occur any more. 😂 
```
I (363) esp_image: segment 2: paddr=0013abf0 vaddr=4212abf0 size=287d4h (165844) map
I (397) esp_image: segment 3: paddr=001633cc vaddr=40807ca8 size=013f4h (  5108) load
I (399) esp_image: segment 4: paddr=001647c8 vaddr=408090a0 size=01e68h (  7784) load
I (405) esp_image: segment 5: paddr=00166638 vaddr=408496c0 size=002c4h (   708) load
I (414) boot: Loaded app from partition at offset 0x10000
I (418) boot: Disabling RNG early entropy source...
Hello world!
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 200000, tick_rate_hz: 50, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
WARN - esp_wifi_internal_tx 12290
Start connection task. Device capabilities: Ok(EnumSet(Client))
Starting wifi
Wifi started!
About to connect...
```

My guess is that the compiler doesn't watch the cfg.toml file in some case.

bjoernQ, 2024-01-03:
Great to hear it's working! 🎉
I also guess that changes to cfg.toml are not always picked up

closed 2024-01-03 by bjoernQ