esp-rs/esp-wifi-sys #409: BLE example is not work for esp32s3

Volkalex28, 2024-01-11:
Hello. I tried to run an example for Embassy Ble on ESP32S3
I received this output, after which I cannot connect to the device
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c15
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1994
load:0x403c9700,len:0xedc
load:0x403cc700,len:0x2fe0
entry 0x403c9958
I (24) boot: ESP-IDF 439a709c 2nd stage bootloader
I (25) boot: compile time Nov 24 2022 01:40:48
I (25) boot: chip revision: v0.1
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 16MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs_factory      WiFi data        01 02 00011000 00003000
I (66) boot:  1 nvs              WiFi data        01 02 00014000 00019000
I (73) boot:  2 otadata          OTA data         01 00 0002d000 00002000
I (81) boot:  3 phy_init         RF data          01 01 0002f000 00001000
I (88) boot:  4 factory          factory app      00 00 00030000 00400000
I (96) boot:  5 ota_0            OTA app          00 10 00430000 00580000
I (103) boot:  6 ota_1            OTA app          00 11 009b0000 00580000
I (111) boot: End of partition table
I (115) boot: Defaulting to factory image
I (120) esp_image: segment 0: paddr=00030020 vaddr=3c060020 size=12154h ( 74068) map
I (141) esp_image: segment 1: paddr=0004217c vaddr=3fc8edf0 size=0146ch (  5228) load
I (143) esp_image: segment 2: paddr=000435f0 vaddr=3fcabd88 size=002c4h (   708) load
I (147) esp_image: segment 3: paddr=000438bc vaddr=40378000 size=06dech ( 28140) load
I (162) esp_image: segment 4: paddr=0004a6b0 vaddr=00000000 size=05968h ( 22888) 
I (168) esp_image: segment 5: paddr=00050020 vaddr=42000020 size=51854h (333908) map
I (235) boot: Loaded app from partition at offset 0x30000
I (235) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
DEBUG - call coex-initialize
INFO - coex firmware version: ebddf30

TRACE - semphr_create - max 1 init 1
TRACE - sem created res = 0 (+1)
TRACE - semphr_create - max 1 init 1
TRACE - sem created res = 1 (+1)
DEBUG - timer_setfn 3fca8c68 0x420219a0 0x0
TRACE - >>>> semphr_take 0x1 block_time_tick 4294967295
TRACE - >>>> return from semphr_take
INFO - coexist rom version e7ae62f

TRACE - alloc 48
TRACE - semphr_give 0x1
DEBUG - ble init
DEBUG - g_bt_plf_log_level = 1
btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010006
DEBUG - coex-init
DEBUG - BT controller compile version 85b425c
TRACE - alloc 280
TRACE - alloc 976
TRACE - alloc 2344
MAGIC fadebead VERSION 00010001
TRACE - alloc 40
TRACE - alloc 284
TRACE - alloc 220
TRACE - alloc 680
TRACE - alloc 608
TRACE - alloc 444
TRACE - alloc 260
TRACE - alloc 3056
TRACE - alloc 11436
TRACE - alloc 908
RWIP Heap alloc: ENV [0x3fc93a90 3056], MSG [0x3fc94684 11436], NORET [0x3fc97334 908], DB [0x00000000 0]
TRACE - alloc 12
Uart ENV [0x00000000], VHCI ENV [0x3fc976c4]
TRACE - semphr_create - max 1 init 0
TRACE - sem created res = 2 (+1)
TRACE - semphr_create - max 1 init 0
TRACE - sem created res = 3 (+1)
TRACE - semphr_create - max 1 init 1
TRACE - sem created res = 4 (+1)
TRACE - task_create 0x4037bf20 0x3c069a2c btController 8192 0x0 200 0x3fceff9c 0
DEBUG - spawning task btController: 0x4037bf20 param 0x0 prio 200
TRACE - queue posting [7, 0, 0, 0, 0, 0, 0, 0]
DEBUG - task started: 0x4037bf20 0x0
TRACE - >>>> semphr_take 0x4 block_time_tick 4294967295
TRACE - semphr_give 0x4
TRACE - >>>> semphr_take 0x3 block_time_tick 10000
TRACE - >>>> return from semphr_take
TRACE - queue_recv 0x3fc91f28 item 0x3fca41b0 block_time_tick 0TRACE - >>>> return from semphr_take with timeout
DEBUG - The btdm_controller_init was initialized
TRACE - received [7, 0, 0, 0, 0, 0, 0, 0]
TRACE - queue_recv returns
phase match cali!!
hi2, lo2
TRACE - >>>> semphr_take 0x1 block_time_tick 4294967295
TRACE - >>>> return from semphr_take

TRACE - read_mac 0x3fcef9e8 2
BD_ADDR: F6:12:FA:D5:40:C1
EM_ET_OFFSET: 0x00000000, EM_ET_END: 0x0000010TRACE - semphr_give 0x1
TRACE - phy_enable_clock
TRACE - phy_enable_clock done!
```

Volkalex28, 2024-01-11:
I think there is a hang on this function: `register_chipv7_phy`

bjoernQ, 2024-01-11:
Which chip / dev-board is that exactly? Haven't had any problems with DevKitC-1 1.6 or the original S3-Box
Does it work fine with esp-idf examples (e.g. https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph )

Volkalex28, 2024-01-11:
This is a custom board. ble with idf has been running successfully on it for a year now
I ran this example: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/examples/embassy_ble.rs

bjoernQ, 2024-01-11:
What is the output of `espflash board-info`?

e.g.
```
❯ espflash board-info
[2024-01-11T09:12:35Z INFO ] Serial port: 'COM20'
[2024-01-11T09:12:35Z INFO ] Connecting...
[2024-01-11T09:12:35Z INFO ] Using flash stub
[2024-01-11T09:12:37Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       xx:xx:xx:xx:xx:xx
```

Volkalex28, 2024-01-11:
```
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       f4:12:fa:d5:40:c0
```

bjoernQ, 2024-01-11:
Weird - basically same. When it hangs in `register_chipv7_phy` it usually means that BLE/WiFi clocks are not enabled

Volkalex28, 2024-01-11:
But I ran the code from the example)

bjoernQ, 2024-01-11:
Yes - it definitely shouldn't happen

Here is how it should look like
```
❯ pwd
C:\projects\clean\esp-wifi\esp-wifi

esp-wifi/esp-wifi on  main is 📦 v0.2.0 via 🦀 v1.75.0-nightly
❯ cargo +esp --version
cargo 1.75.0-nightly (1d8b05cdd 2023-11-20)

esp-wifi/esp-wifi on  main is 📦 v0.2.0 via 🦀 v1.75.0-nightly
❯ cargo +esp esp32s3 --release --features=ble --example=ble
   Compiling compiler_builtins v0.1.103
   Compiling core v0.0.0 (C:\Users\Bjoern\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core)
   Compiling proc-macro2 v1.0.71
   Compiling unicode-ident v1.0.12
   ....
   Compiling esp-wifi v0.2.0 (C:\projects\clean\esp-wifi\esp-wifi)
   Compiling xtensa-lx-rt v0.16.0
    Finished release [optimized + debuginfo] target(s) in 47.13s
     Running `espflash flash --monitor C:\projects\clean\esp-wifi\target\xtensa-esp32s3-none-elf\release\examples\ble`
[2024-01-11T10:15:50Z INFO ] Serial port: 'COM20'
[2024-01-11T10:15:50Z INFO ] Connecting...
[2024-01-11T10:15:50Z INFO ] Using flash stub
[2024-01-11T10:15:52Z WARN ] Setting baud rate higher than 115,200 can cause issues
Chip type:         esp32s3 (revision v0.1)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       68:b6:b3:4d:ae:08
App/part. size:    454,544/8,323,072 bytes, 5.46%
[00:00:06] [========================================]     281/281     0x10000
[2024-01-11T10:16:00Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - pp_sig_cnt
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - hal_pwr_interrupt_clr_event
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - hal_pwr_interrupt_clr_event
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - hal_pwr_interrupt_clr_event
    at ??:??
I (48) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (48) boot: compile time Jun  7 2023 08:07:32
I (49) boot: Multicore bootloader
I (54) boot: chip revision: v0.1
I (57) boot.esp32s3: Boot SPI Speed : 40MHz
I (62) boot.esp32s3: SPI Mode       : DIO
I (67) boot.esp32s3: SPI Flash Size : 8MB
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (95) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (103) boot:  2 factory          factory app      00 00 00010000 007f0000
I (110) boot: End of partition table
I (115) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=119f8h ( 72184) map
I (141) esp_image: segment 1: paddr=00021a20 vaddr=3fc8f1f0 size=00fc4h (  4036) load
I (143) esp_image: segment 2: paddr=000229ec vaddr=3fca8eec size=00198h (   408) load
I (147) esp_image: segment 3: paddr=00022b8c vaddr=40378000 size=071ech ( 29164) load
I (164) esp_image: segment 4: paddr=00029d80 vaddr=00000000 size=06298h ( 25240)
I (171) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=4ef44h (323396) map
I (255) boot: Loaded app from partition at offset 0x10000
I (255) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
Ok(CommandComplete { num_packets: 5, opcode: 3075, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8198, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8200, data: [0] })
Ok(CommandComplete { num_packets: 5, opcode: 8202, data: [0] })
started advertising

esp-wifi/esp-wifi on  main is 📦 v0.2.0 via 🦀 v1.75.0-nightly
```

Volkalex28, 2024-01-11:
I decided to check what would happen if I disable the coex feature. The same

Volkalex28, 2024-01-15:
I did a long series of experiments and found that enabling the "phy-enable-usb" feature solved this problem

Volkalex28, 2024-01-15:
What is this anyway and why?

bjoernQ, 2024-01-15:
Ah you connected the S3 via serial-jtag?
By default enabling the wifi-ble-phy will disable the USB phy. The feature prevents that

However the device should advertise and be connectible without -  just not outputting anything after initialization and you might need to force it manually into download mode.

But great you got it to work

Volkalex28, 2024-01-15:
Cool! Thank you

closed 2024-01-15 by Volkalex28