esp-rs/esp-wifi-sys #426: esp-wifi 0.3.0 does not compile for ESP32-C3, it depends on portable-atomic, and that doesnt build for RiscV

makepaddev, 2024-01-31:
I can only build esp-wifi 0.1.1 for the ESP32-C3 (riscV), so this means i can't try the newer releases.
The 0.1.1 version has significant problems on ESP32-C3 with wifi connectivity, UDP packets arrive super slowly or not at all. A quick check with the C++ stack on arduino IDE show its not a hardware issue as it works perfectly there.

Trying to compile with 0.3.0 gives this error: 

error: cfg(portable_atomic_unsafe_assume_single_core) does not compatible with this target;
       if you need cfg(portable_atomic_unsafe_assume_single_core) support for this target,
       please submit an issue at <https://github.com/taiki-e/portable-atomic>
   --> /Users/admin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.6.0/src/lib.rs:362:1
    |
362 | / compile_error!(
363 | |     "cfg(portable_atomic_unsafe_assume_single_core) does not compatible with this target;\n\
364 | |      if you need cfg(portable_atomic_unsafe_assume_single_core) support for this target,\n\
365 | |      please submit an issue at <https://github.com/taiki-e/portable-atomic>"
366 | | );
    | |_^

makepaddev, 2024-01-31:
I was able to fix the atomic error thanks to https://github.com/taiki-e/portable-atomic/issues/148, however no i get this when compiling esp-wifi 0.3.0

error[E0432]: unresolved import `crate::hal::interrupt`
 --> /Users/admin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.3.0/src/preempt/preempt_riscv.rs:3:17
  |
3 | use crate::hal::interrupt::TrapFrame;
  |                 ^^^^^^^^^ could not find `interrupt` in `hal`

error[E0432]: unresolved imports `crate::hal::interrupt`, `crate::hal::macros::interrupt`
   --> /Users/admin/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.3.0/src/timer/timer_esp32c3.rs:5:11
    |
5   |     hal::{interrupt, macros::interrupt, peripherals::Interrupt},
    |           ^^^^^^^^^  ^^^^^^^^^^^^^^^^^ no `interrupt` in the root
    |           |
    |           no `interrupt` in the root

bjoernQ, 2024-02-01:
esp-wifi releases are tied to esp-hal releases - they need to match! e.g. for 0.3.0 the C3 hal is 0.15.0: https://crates.io/crates/esp-wifi/0.3.0/dependencies

additionally the way how atomics are handled in esp-hal changed at some point in time - for later HAL versions atomic emulation was removed so you need to remove this: https://github.com/makepaddev/rc_radio/blob/a04fd8eec7278f7d924d11488b25bf384c249050/.cargo/config.toml#L13-L27

makepaddev, 2024-02-01:
Hi! Thanks for responding, so just for clean-ness im just trying to run the examples in esp-wifi from this repo intead.
I have the ESP32-C3 SEEED XIAO boards.
Im calling this (rustup of today):
cargo +nightly esp32c3 --example dhcp --release --features "wifi"

I have a few warnings like this:
warning: shared reference of mutable static is discouraged
  --> esp-wifi/src/compat/common.rs:37:67

But otherwise it compiles, then flashes, then i get this:
Any hints what i did wrong? Just off the bat it seems like an esp32s3 rom is not what you would need for a c3? as it says a the top:

[2024-02-01T11:30:37Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c72
0x40378c72 - bars
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - bars
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - hal_mac_txq_disable
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - hal_mac_txq_disable
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - hal_mac_txq_disable
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...
entry 0x403c9908
0x403c9908 - hal_mac_txq_disable
    at ??:??
I (45) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (46) boot: compile time Jun  7 2023 08:07:32
I (47) boot: Multicore bootloader
I (51) boot: chip revision: v0.2
I (55) boot.esp32s3: Boot SPI Speed : 40MHz
I (59) boot.esp32s3: SPI Mode       : DIO
I (64) boot.esp32s3: SPI Flash Size : 8MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (93) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (100) boot:  2 factory          factory app      00 00 00010000 007f0000
I (108) boot: End of partition table
I (112) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=11f10h ( 73488) map
I (139) esp_image: segment 1: paddr=00021f38 vaddr=3fc83e28 size=010b0h (  4272) load
E (139) esp_image: Segment 1 0x3fc83e28-0x3fc84ed8 invalid: bad load address range
0x3fc83e28 - g_wifi_feature_caps
    at ??:??
0x3fc84ed8 - _ZN8esp_wifi4wifi16WIFI_TX_INFLIGHT17h25c9de066e770dbeE
    at ??:??
E (144) boot: Factory app partition is not bootable
E (150) boot: No bootable app partitions in the partition table
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403cdd51
0x403cdd51 - hal_mac_txq_disable
    at ??:??
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
0x3fce3818 - bars
    at ??:??
load:0x403c9700,len:0x4
0x403c9700 - hal_mac_txq_disable
    at ??:??
load:0x403c9704,len:0xc00
0x403c9704 - hal_mac_txq_disable
    at ??:??
load:0x403cc700,len:0x2eb0
0x403cc700 - hal_mac_txq_disable
    at ??:??
SHA-256 comparison failed:
Calculated: b8f485cf73c8ad59b009a332167b4a854ec774585c7fa428926acc30cc0c964b
Expected: 3499e4149a363c2b1ee4ee9709ca5031f50ee6b595a532f697c8659516763ae3
Attempting to boot anyway...

MabezDev, 2024-02-01:
That message about the esp32s3 is coming from ROM code, which is burned into flash at the factory - this means you have an esp32s3 board, not a esp32c3 board. Try flashing the example targetting the esp32s3.

makepaddev, 2024-02-01:
Sorry im mistaken, i had an S3 board plugged in, nevermind.

makepaddev, 2024-02-01:
Alright it works! I have a working example. Thanks. Closing.

closed 2024-02-01 by makepaddev