esp-rs/esp-wifi-sys #427: How to depend on esp-wifi in an external example

makepaddev, 2024-02-01:
Hi!,

After getting the esp-wifi examples to work, i'm wondering how i should depend on esp-wifi for an external project.
There are so many features flags / configs going on in the examples i have no idea how to extract it. Ideally it uses the same 'alias' system to switch chips, but i'm also fine hardcoding it somehow. I keep getting compiler errors because im probably missing some feature flag somewhere. Is there an example of an external project depending on esp-wifi? Also does esp-wifi 'forward' things like the hal and its other dependencies? Having to manually manage the versions of these in the end crate is also a headache.

Thanks

makepaddev, 2024-02-01:
Say i want to put the DHCP example in an external project, how would that work?

bjoernQ, 2024-02-01:
We have the features documented here: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/README.md#features (plus the feature for the chip you are targeting)

You could look at repositories using esp-wifi: https://github.com/esp-rs/esp-wifi/network/dependents - but probably not all of them are updated recently

We don't re-export esp-hal or any other of our dependencies

MabezDev, 2024-02-20:
@makepaddev [esp-template](https://github.com/esp-rs/esp-template) supports some level of code generation for esp-wifi, you can then cross reference the example features to build up to the DHCP example.

closed 2024-02-20 by MabezDev