esp-rs/esp-wifi-sys #431: Long async ping time when idle

karlri, 2024-02-10:
I tested the async performance and it is great. There is one small issue with it though. When async code is idle (no throughput) the ping time becomes long because there are no requests for tx_token or rx_token that cause task switching.

I made a hacky modification to `esp-hal-common-0.15.0/src/embassy/executor/thread.rs` that seems to resolve the issue. It seems to push the idle ping time down to about ~2ms even with tick_rate_hz = 1. With the hack, the task switching frequency remains quite low, but we react faster to incoming wifi data.

```Rust

    #[cfg(xtensa)]
    pub fn wait_impl(cpu: usize) {
        // Wait until we are ready to re-poll some future.
        while Self::wait_once(cpu) {}
    }

    #[cfg(xtensa)]
    /// Returns true if we need to wait again.
    pub fn wait_once(cpu: usize) -> bool {
        // If we're using esp-wifi and we're running on core 0 and we're not ready
        // we might become ready faster by task switching to the wifi driver.
        if cpu == 0 && !SIGNAL_WORK_THREAD_MODE[cpu].load(Ordering::SeqCst) {
            fn yield_task() {
                let intr = 1 << 29;
                unsafe {
                    core::arch::asm!("wsr.intset  {0}", in(reg) intr, options(nostack));
                }
            }
            yield_task();
        }
        // Manual critical section implementation that only masks interrupts handlers.
        // We must not acquire the cross-core on dual-core systems because that would
        // prevent the other core from doing useful work while this core is sleeping.
        let token: critical_section::RawRestoreState;
        unsafe { core::arch::asm!("rsil {0}, 5", out(reg) token) };

        // we do not care about race conditions between the load and store operations,
        // interrupts will only set this value to true.
        if SIGNAL_WORK_THREAD_MODE[cpu].load(Ordering::SeqCst) {
            SIGNAL_WORK_THREAD_MODE[cpu].store(false, Ordering::SeqCst);

            // if there is work to do, exit critical section and loop back to polling
            unsafe {
                core::arch::asm!(
                    "wsr.ps {0}",
                    "rsync",
                    in(reg) token
                );
            }

            false // we don't need to wait again.
        } else {
            // waiti sets the PS.INTLEVEL when slipping into sleep
            // because critical sections in Xtensa are implemented via increasing
            // PS.INTLEVEL the critical section ends here
            // take care not add code after `waiti` if it needs to be inside the CS
            unsafe { core::arch::asm!("waiti 0") }; /* critical section ends here */

            true // we need to wait again. We don't know if the interrupt made us ready or not.
        }
    }
```

I tried putting the task switching after `waiti` but that causes a storm of interrupts so that seems no good. I don't understand why.

Without the above hack:
```
tick_rate_hz = 100
64 bytes from 192.168.68.112: icmp_seq=9450 ttl=64 time=12.6 ms
64 bytes from 192.168.68.112: icmp_seq=9451 ttl=64 time=12.5 ms
64 bytes from 192.168.68.112: icmp_seq=9452 ttl=64 time=12.2 ms
64 bytes from 192.168.68.112: icmp_seq=9453 ttl=64 time=12.1 ms
```

With the above hack:
```
tick_rate_hz = 1
64 bytes from 192.168.68.112: icmp_seq=9140 ttl=64 time=2.32 ms
64 bytes from 192.168.68.112: icmp_seq=9141 ttl=64 time=2.20 ms
64 bytes from 192.168.68.112: icmp_seq=9142 ttl=64 time=2.33 ms
64 bytes from 192.168.68.112: icmp_seq=9143 ttl=64 time=2.10 ms
```

What do you think about the general idea? Any tips for how to resolve it in a better way?

karlri, 2024-02-10:
Also, it seems to speed up connecting to wifi and obtaining an IP with dhcpv4.

MabezDev, 2024-02-12:
Out of curiosity, if you up the tick rate to say, 1000hz, do you achieve similar results?

I think to do this properly we'd either need to modify the executor in esp-hal directly, or add some kind of idle hook which can choose not to sleep, which would be a reasonable amount of work. If upping the hz yields the same result I'd be inclined to just do that for the latency-sensitive applications.

karlri, 2024-02-12:
Upping the task switching frequency helps, but it's not enough to get the same kind of responsiveness. Here are some new tests. It's in the middle of the workday and I'm working from home so very low interference and noise at the moment so the results are even better now.

with the above hack and `tick_rate_hz=1`
```
PING 192.168.68.105 (192.168.68.105) 56(84) bytes of data.
64 bytes from 192.168.68.105: icmp_seq=1 ttl=64 time=1.84 ms
64 bytes from 192.168.68.105: icmp_seq=2 ttl=64 time=1.81 ms
64 bytes from 192.168.68.105: icmp_seq=3 ttl=64 time=1.84 ms
64 bytes from 192.168.68.105: icmp_seq=4 ttl=64 time=1.91 ms
64 bytes from 192.168.68.105: icmp_seq=5 ttl=64 time=1.89 ms
64 bytes from 192.168.68.105: icmp_seq=6 ttl=64 time=1.88 ms
```

no hack and `tick_rate_hz=1000`
```
PING 192.168.68.105 (192.168.68.105) 56(84) bytes of data.
64 bytes from 192.168.68.105: icmp_seq=1 ttl=64 time=4.24 ms
64 bytes from 192.168.68.105: icmp_seq=2 ttl=64 time=4.44 ms
64 bytes from 192.168.68.105: icmp_seq=3 ttl=64 time=3.76 ms
64 bytes from 192.168.68.105: icmp_seq=4 ttl=64 time=4.23 ms
64 bytes from 192.168.68.105: icmp_seq=5 ttl=64 time=3.00 ms
```

The hook approach would be nice, but that might affect several crates I suppose.