esp-rs/esp-wifi-sys #436: example/embassy_ble does not compile

alexkazik, 2024-03-06:
I've tried to compile (mainline Rust compiler) but it doesn't work, and I don't know how to fix it.
`cargo +nightly esp32c3 --example embassy_ble --release --features "async,ble"`

```
error: async closure does not implement `FnMut` because it captures state from its environment
   --> esp-wifi/examples/embassy_ble.rs:147:28
    |
147 |         let mut notifier = async || {
    |                            ^^^^^^^^
...
162 |         srv.run(&mut notifier).await.unwrap();
    |             --- required by a bound introduced by this call
    |
```

And other examples seem also not to work.

bjoernQ, 2024-03-06:
For recent compiler versions there is a small change needed. It's already prepared in a PR here: https://github.com/esp-rs/esp-wifi/pull/434/files#diff-4fa6b3b3d80c35ebc34a9b33a487051271cb99f4a03f46fcff852f553e764547 but that PR is waiting for the upcoming release of esp-hal (which will be soon)

With the changes in the PR it should work - let me know if it doesn't

alexkazik, 2024-03-06:
Yes, that does compile (though I currently can't test it on hardware).

closed 2024-03-12 by alexkazik