esp-rs/esp-wifi-sys #438: multiple definition of `malloc' `calloc` and `free`

noahbliss, 2024-03-12:
Hey! I use another crate which wraps some C code. That crate and esp-wifi both seem to rely on the C functions listed in the title. When both crates are used, the compiler dies of an aneurysm. Any ideas on possible solutions? Could this be added as a feature perhaps where a user can bring their own implementations of these functions? I'll request the same behavior from the other crate's maintainer. 

Very simply, using the [a121-rs](https://github.com/Ragarnoy/a121-rs/) crate and esp-wifi together doesn't work.

noahbliss, 2024-03-12:
My noobish recommendation would be to make a default-enabled feature which enables these functions inside the crate, but that users can opt-out of if they have implemented them inside their own parent project.

bjoernQ, 2024-03-12:
Just not including those functions is technically possible but where does the other crate place the heap? How does the heap get configured? The wifi drivers need a good chunk of memory (by default we reserve 64k but for some configurations more is needed)