esp-rs/esp-wifi-sys #441: [ESP32] esp-wifi crashes when trying to connect via BLE

thecodechemist99, 2024-03-14:
# Context

I’m working on a project using embassy and the esp-wifi crate for BLE.
I use a Lolin32-lite board with an ESP32 Rev1.

# Unexpected Behaviour

When I try to connect to the ESP32 from my Android phone, the ESP32 crashes with the following error message:

```
!! A panic occured in '/Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/preempt/preempt_xtensa.rs', at line 104, column 9:
index out of bounds: the len is 3 but the index is 1073503776

Backtrace:

0x40122d09
0x40122d09 - core::panicking::panic_bounds_check
    at /Users/thecodechemist99/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/panicking.rs:208
0x400ea9a1
0x400ea9a1 - esp_wifi::preempt::preempt::save_task_context
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/preempt/preempt_xtensa.rs:104
0x400ead30
0x400ead30 - esp_wifi::timer::arch_specific::__esp_hal_internal_TG1_T0_LEVEL
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/timer/xtensa.rs:99
0x40081510
0x40081510 - esp_hal::interrupt::xtensa::vectored::handle_interrupt
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:452
0x400810e2
0x400810e2 - esp_hal::interrupt::xtensa::vectored::handle_interrupts
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:432
0x40080dc6
0x40080dc6 - __level_2_interrupt
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:376
0x40081817
0x40081817 - __default_naked_level_2_interrupt
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.16.0/src/exception/assembly_esp32.rs:599
0x4003ffff
0x4003ffff - r_ld_sched_sniff_add
    at ??:??
0x400fd4a3
0x400fd4a3 - esp_hal::critical_section_impl::xtensa::<impl critical_section::Impl for esp_hal::critical_section_impl::CriticalSection>::release
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/lib.rs:279
0x4010e2da
0x4010e2da - critical_section::release
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:200
```

# Code Example

Here’s my code to reproduce the issue:

```Rust
#![no_std]
#![no_main]
#![feature(async_closure)]
#![feature(type_alias_impl_trait)]

// Logging traits
use esp_backtrace as _;
use esp_println::println;

// Core traits
use embassy_executor::Spawner;
use embassy_sync::{blocking_mutex::raw::CriticalSectionRawMutex, channel::Channel};
use embassy_time::Timer;
use esp_hal::{
    adc::{AdcConfig, AdcPin, Attenuation, ADC},
    clock::ClockControl,
    embassy,
    gpio::{Analog, GpioPin},
    peripherals::*,
    prelude::*,
    timer::TimerGroup,
    Rng, IO,
};

// BLE traits
use bleps::{
    ad_structure::{
        create_advertising_data, AdStructure, BR_EDR_NOT_SUPPORTED, LE_GENERAL_DISCOVERABLE,
    },
    async_attribute_server::AttributeServer,
    asynch::Ble,
    attribute_server::NotificationData,
    gatt,
};
use esp_wifi::{
    ble::controller::asynch::BleConnector, initialize, EspWifiInitFor, EspWifiInitialization,
};

// Message channels
static SENSOR_READING: Channel<CriticalSectionRawMutex, [u8; 4], 1> = Channel::new();

#[main]
async fn main(spawner: Spawner) {
    // Set up system
    let peripherals = Peripherals::take();
    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();
    let timg0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    embassy::init(&clocks, timg0);

    // Set up IO
    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);

    // Set up BLE
    let timer = TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
    let init = initialize(
        EspWifiInitFor::Ble,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();

    let bluetooth = peripherals.BT;

    // Set up ADC
    let mut adc_config = AdcConfig::new();
    let adc_pin = adc_config.enable_pin(io.pins.gpio35.into_analog(), Attenuation::Attenuation0dB);
    let adc = ADC::<ADC1>::new(peripherals.ADC1, adc_config);

    // Spawn tasks
    spawner.spawn(ble_runner(init, bluetooth)).unwrap();
    spawner.spawn(read_sensor(adc, adc_pin)).unwrap();
}

/// Ble runner task
#[embassy_executor::task]
async fn ble_runner(init: EspWifiInitialization, mut bluetooth: BT) {
    let connector = BleConnector::new(&init, &mut bluetooth);
    let mut ble = Ble::new(connector, esp_wifi::current_millis);

    loop {
        // Configure advertising
        println!("{:?}", ble.init().await);
        println!("{:?}", ble.cmd_set_le_advertising_parameters().await);
        println!(
            "{:?}",
            ble.cmd_set_le_advertising_data(
                create_advertising_data(&[
                    AdStructure::Flags(LE_GENERAL_DISCOVERABLE | BR_EDR_NOT_SUPPORTED),
                    AdStructure::ServiceUuids128(&[
                        Uuid::Uuid128([
                            0x8d, 0xa4, 0xc3, 0xd7, 0x0, 0x0, 0x4b, 0x26, 0x96, 0x32, 0xae, 0x20,
                            0x7b, 0x6b, 0x75, 0x7f,
                        ]), // Custom Sensor Service
                    ]),
                    AdStructure::CompleteLocalName("ESP32"),
                ])
                .unwrap(),
            )
            .await
        );
        println!("{:?}", ble.cmd_set_le_advertise_enable(true).await);

        println!("started advertising");

        // Define service methods
        let mut read_sensor = |_offset: usize, data: &mut [u8]| {
            // Try to get last reading, otherwise send 0.0
            if let Ok(value) = SENSOR_READING.try_receive() {
                data.copy_from_slice(&value);
            } else {
                data.copy_from_slice(&[0u8; 4]);
            }
            4
        };

        gatt!([
            // Custom Sensor Service
            service {
                uuid: "8da4c3d7-0000-4b26-9632-ae207b6b757f",
                characteristics: [
                    // Normalized sensor reading
                    characteristic {
                        name: "sensor_reading",
                        uuid: "8da4c3d7-4514-0001-9632-ae207b6b757f",
                        notify: true,
                        read: read_sensor,
                    }
                ]
            }
        ]);

        // Set up attribute server
        let mut rng = bleps::no_rng::NoRng;
        let mut srv = AttributeServer::new(&mut ble, &mut gatt_attributes, &mut rng);

        let mut notifier = async || {
            let mut data = [0u8; 4];
            data.copy_from_slice(&SENSOR_READING.receive().await);
            NotificationData::new(sensor_reading_handle, &SENSOR_READING.receive().await)
        };

        // Start server
        if let Some(e) = srv.run(&mut notifier).await.err() {
            println!("gatt_server run exited with error: {:?}", e);
        }
    }
}

/// Read sensor task
#[embassy_executor::task]
async fn read_sensor(mut adc: ADC<'static, ADC1>, mut pin: AdcPin<GpioPin<Analog, 35>, ADC1>) {
    loop {
        let raw_value: u32 = nb::block!(adc.read(&mut pin)).unwrap() as u32;
        let value: f32 = normalize_adc_reading(raw_value, true);
        let bytes = value.to_be_bytes();

        // Send bytes to ble runner
        if let Ok(_) = SENSOR_READING.try_send(bytes) {
            println!("{:?}", value);
        };

        // Re-schedule the timer interrupt in 100ms
        Timer::after_millis(100).await;
    }
}

fn normalize_adc_reading(mut rdg: u32, inverted: bool) -> f32 {
    if inverted {
        rdg = 4095 - rdg;
    }

    rdg as f32 / 4096.0
}
```

thecodechemist99, 2024-03-14:
I have also encountered these additional errors, but cannot reliably reproduce them for now:

```
!! A panic occured in '/Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/ble/btdm.rs', at line 195, column 51:
already borrowed: BorrowMutError

Backtrace:

0x401297fd
0x401297fd - core::cell::panic_already_borrowed
    at /Users/thecodechemist99/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/cell.rs:761
0x400808c8
0x400808c8 - core::cell::RefCell<T>::borrow_mut
    at /Users/thecodechemist99/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/cell.rs:1051
0x400809a4
0x400809a4 - esp_wifi::ble::btdm::queue_send_from_isr
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/ble/btdm.rs:215
0x4008a7ef
0x4008a7ef - btdm_task_post_from_isr
    at ??:??
0x4008abd1
0x4008abd1 - r_rwbtdm_isr_wrapper
    at ??:??
0x400ea55a
0x400ea55a - esp_wifi::timer::chip_specific::__esp_hal_internal_RWBLE
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/timer/timer_esp32.rs:77
0x40081510
0x40081510 - esp_hal::interrupt::xtensa::vectored::handle_interrupt
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:452
0x400810e2
0x400810e2 - esp_hal::interrupt::xtensa::vectored::handle_interrupts
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:432
0x40080db2
0x40080db2 - __level_1_interrupt
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-hal-0.16.1/src/interrupt/xtensa.rs:370
0x4008176b
0x4008176b - __default_naked_exception
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/xtensa-lx-rt-0.16.0/src/exception/assembly_esp32.rs:479
```

```
Exception occured 'InstrProhibited'
Context
PC=0x00000001       PS=0x00060011
0x00000001 - esp32::AES::steal
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1521
0x00060011 - PS_WOE
    at ??:??
A0=0x800814ba       A1=0x3ffc3930       A2=0x00000001       A3=0x3ffc3ce0       A4=0x8010e20d
0x800814ba - _rtc_slow_data_end
    at ??:??
0x3ffc3930 - _ZN8esp_wifi7preempt10TASK_STACK17hb41e634bda03073bE
    at ??:??
0x00000001 - esp32::AES::steal
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1521
0x3ffc3ce0 - _ZN8esp_wifi7preempt10TASK_STACK17hb41e634bda03073bE
    at ??:??
0x8010e20d - _rtc_slow_data_end
    at ??:??
A5=0x3ffc3a30       A6=0x00060700       A7=0x00000014       A8=0x800ea4ad       A9=0x00000004
0x3ffc3a30 - _ZN8esp_wifi7preempt10TASK_STACK17hb41e634bda03073bE
    at ??:??
0x00060700 - PS_WOE
    at ??:??
0x00000014 - XT_STK_A3
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:2037
0x800ea4ad - _rtc_slow_data_end
    at ??:??
0x00000004 - esp32::AES::steal
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp32-0.29.0/src/lib.rs:1521
[... 105 lines cut ...]
0x40040001
0x40040001 - r_ld_sched_sniff_add
    at ??:??
0x40080a1f
0x40080a1f - esp_wifi::ble::btdm::cause_sw_intr_to_core
    at /Users/thecodechemist99/.cargo/registry/src/index.crates.io-6f17d22bba15001f/esp-wifi-0.4.0/src/ble/btdm.rs:337
0x401506ab
0x401506ab - vhci_send
    at ??:??
0x4013d895
0x4013d895 - r_eif_send
    at ??:??
0x40017920
0x40017920 - ecc_point_multiplication_uint8_256
    at ??:??
0x4013e244
0x4013e244 - hci_tx_start
    at ??:??
0x4013e321
0x4013e321 - r_hci_tl_send
    at ??:??
0x400188f8
0x400188f8 - r_hci_send_2_host
    at ??:??
```

bjoernQ, 2024-03-14:
Can you run the provided examples without issues?

both of the crashes sound a lot like stack or general memory corruption to me

thecodechemist99, 2024-03-14:
> Can you run the provided examples without issues?

The ble examples work fine. I also had all parts of my own code working before, reading the sensor and connecting via bluetooth. But when I try to put it all together, I end up with these errors.

> both of the crashes sound a lot like stack or general memory corruption to me
If that would be the case, what could I do about it?

bjoernQ, 2024-03-15:
Ok looking again there is one obvious thing in the code - nothing weird and explainable.

```rust
 // Define service methods
        let mut read_sensor = |_offset: usize, data: &mut [u8]| {
            // Try to get last reading, otherwise send 0.0
            if let Ok(value) = SENSOR_READING.try_receive() {
                data.copy_from_slice(&value);
            } else {
                data.copy_from_slice(&[0u8; 4]);
            }
            4
        };
```

The `data` slice is 255 bytes so the correct thing would be
```rust
        // Define service methods
        let mut read_sensor = |_offset: usize, data: &mut [u8]| {
            // Try to get last reading, otherwise send 0.0
            if let Ok(value) = SENSOR_READING.try_receive() {
                data[..4].copy_from_slice(&value);
            } else {
                data[..4].copy_from_slice(&[0u8; 4]);
            }
            4
        };
```

thecodechemist99, 2024-03-15:
@bjoernQ thanks, I already found and fixed that yesterday. Didn‘t notice it was still in here.

bjoernQ, 2024-03-15:
I guess we can close this issue now

closed 2024-03-15 by bjoernQ