esp-rs/esp-wifi-sys #445: ps-min-modem / ps-max-modem not working on esp32c3

rbomze, 2024-03-22:
Hi,
today i extensively experimented with esp-rs in combination with esp-wifi and embassy-rs on the esp32 platform.
On a esp32c3 everything works fine as long there is no modem sleep enabled. Enabling eg. ps-min-modem via a feature for esp-wifi, allows to connect to an AP without any problems, but any further connection attempts via socket fail. 
Not a single packet is received on the other side (i am getting 'ConnectionResets' as a fault).

Curious, because i was always using modem sleep in all esp-idf projects before (on the 'classic' esp32, where it is the default  btw.) i tried to run (almost) the identical code on a 'classic' esp32. This time 'ps-min-modem' worked without any problems.
The only real difference between both codes are the timers:
```
    #[cfg(target_arch = "riscv32")]
    let timer = esp_hal::systimer::SystemTimer::new(peripherals.SYSTIMER).alarm0;
    #[cfg(target_arch = "xtensa")]
    let timer = esp_hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
```
Could it be that the alarm0 timer is kind of too inaccurate to handle the right timing of the beacons and DTIMs? Btw. i tried different intervals for the the beacons [between 100-500] and DTIMs [between 2-5] - on the esp32c3 nothing worked with modem sleep enabled, on the esp32, as expected  i got  more latency with higher settings.

bjoernQ, 2024-03-25:
I just tried it and both, `ps-min-modem` and `ps-max-modem` work fine for me running on ESP32-C3. (Connecting to Fritz!Box 5590 as AP)

I don't think the 16MHz clock is too inaccurate for the job. I assume besides using a different all the setup is the same?
I assume you are running in `--release` mode? Non-release mode might work but is not recommended/supported.

rbomze, 2024-03-27:
> I just tried it and both, `ps-min-modem` and `ps-max-modem` work fine for me running on ESP32-C3. (Connecting to Fritz!Box 5590 as AP)

I'll try tomorrow with other APs.

> I don't think the 16MHz clock is too inaccurate for the job. I assume besides using a different all the setup is the same? I assume you are running in `--release` mode? Non-release mode might work but is not recommended/supported.

Yes, i was running the code in `--release` mode.
I will investigate further. I'll try the code on the bare hal, without embassy-rs. If nothing helps i pull out my oscilloscope and check the timings.

ps. if the socket communication is done directly after a wifi connect, without any delay, it works. it stops to work if the communication attempts are made after some time idling.