esp-rs/esp-wifi-sys #447: compile error:w

goiw111, 2024-03-25:
hllow guys can you hellp me pleass

# **the code from esp-wifi repo**

 `#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use embassy_executor::Spawner;
use embassy_net::tcp::TcpSocket;
use embassy_net::{Config, Ipv4Address, Stack, StackResources};
use embassy_time::{Duration, Timer};
use esp_backtrace as _;
use esp_println::println;
use esp_wifi::wifi::{ClientConfiguration, Configuration};
use esp_wifi::wifi::{WifiController, WifiDevice, WifiEvent, WifiStaDevice, WifiState};
use esp_wifi::{initialize, EspWifiInitFor};
use esp_hal::clock::ClockControl;
use esp_hal::Rng;
use esp_hal::{embassy, peripherals::Peripherals, prelude::*, timer::TimerGroup};
use static_cell::make_static;

const SSID: &str = env!("SSID");
const PASSWORD: &str = env!("PASSWORD");

#[main]
async fn main(spawner: Spawner) -> ! {
    #[cfg(feature = "log")]
    esp_println::logger::init_logger(log::LevelFilter::Info);

    let peripherals = Peripherals::take();

    let system = peripherals.SYSTEM.split();
    let clocks = ClockControl::max(system.clock_control).freeze();

    let timer = esp_hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks).timer0;
    let init = initialize(
        EspWifiInitFor::Wifi,
        timer,
        Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();

    let wifi = peripherals.WIFI;
    let (wifi_interface, controller) =
        esp_wifi::wifi::new_with_mode(&init, wifi, WifiStaDevice).unwrap();

    let timer_group0 = TimerGroup::new(peripherals.TIMG0, &clocks);
    embassy::init(&clocks, timer_group0);

    let config = Config::dhcpv4(Default::default());

    let seed = 1234; // very random, very secure seed

    // Init network stack
    let stack = &*make_static!(Stack::new(
        wifi_interface,
        config,
        make_static!(StackResources::<3>::new()),
        seed
    ));

    spawner.spawn(connection(controller)).ok();
    spawner.spawn(net_task(&stack)).ok();

    let mut rx_buffer = [0; 4096];
    let mut tx_buffer = [0; 4096];

[... 178 lines cut ...]
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(if_hwctrl.o): in function `ic_set_vif':
          (.text.ic_set_vif+0xe5): undefined reference to `__popcountsi2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: (.text.ic_set_vif+0x13b): undefined reference to `__popcountsi2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(pp.o): in function `ppCalTkipMic':
          (.text.ppCalTkipMic+0x2a): undefined reference to `roundup2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(pp.o):(.wifi0iram.34+0x0): undefined reference to `roundup2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(pp.o): in function `ppCalSubFrameLength':
          (.wifi0iram.34+0x12): undefined reference to `roundup2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(pp.o): in function `ppCheckTxAMPDUlength':
          (.text.ppCheckTxAMPDUlength+0x14): undefined reference to `roundup2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(wdev.o): in function `wDev_IndicateAmpdu':
          (.wifirxiram.25+0x134): undefined reference to `roundup2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libpp.a(wdev.o):(.wifirxiram.25+0x207): more undefined references to `roundup2' follow
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libwpa_supplicant.a(common.c.obj):(.literal.wpa_get_ntp_timestamp+0x8): undefined reference to `__bswapsi2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/sohayl/projects/dron-test/target/xtensa-esp32-none-elf/debug/build/esp-wifi-sys-e279863c8ce60e84/out/libwpa_supplicant.a(common.c.obj): in function `wpa_get_ntp_timestamp':
          /home/bjoern/esp/esp-idf/components/wpa_supplicant/src/utils/common.c:140:(.text.wpa_get_ntp_timestamp+0x1f): undefined reference to `__bswapsi2'
          /home/sohayl/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /home/bjoern/esp/esp-idf/components/wpa_supplicant/src/utils/common.c:141:(.text.wpa_get_ntp_timestamp+0x3d): undefined reference to `__bswapsi2'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `dron-test` (bin "dron-test") due to 1 previous error
`

bjoernQ, 2024-03-25:
Most likely you forgot to add the ROM functions linker script: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/README.md#link-configuration

goiw111, 2024-03-25:
> Most likely you forgot to add the ROM functions linker script: https://github.com/esp-rs/esp-wifi/blob/main/esp-wifi/README.md#link-configuration

thank you it's works

closed 2024-03-25 by bjoernQ

raulqtech, 2024-10-01:
I'm having the same issue, but the provided link gives me a 404 response :/