esp-rs/esp-wifi-sys #450: esp32c6 uses only password while config AuthMethod:None

gintsgints, 2024-03-31:
Steps to reproduce:
* update sid/password for required wifi in https://github.com/gintsgints/rust-iot
* compile project and run.

Result:
At my esp32c6 connection is successeful

Required:
As it is WPA password protected access point, it should work with default WPA2Personal

Note: connection fails with - wrong password or auth method WPA2Personal specified with correct credentials.

bjoernQ, 2024-04-02:
Maybe I'm missing something here but given https://github.com/gintsgints/rust-iot/blob/a54216f9fcf5528f719a4f854e48a0b7d7d7c7b1/src/main.rs#L52-L55

It's expected you set the password to `None` if you want to use `AuthMethod::None`. Probably we should return an error if the configuration doesn't make sense in `set_configuration`

gintsgints, 2024-04-02:
I know that `AuthMethod::None` and password is wrong combination, but this is only combination it's work for me.

gintsgints, 2024-04-06:
When I execute code from example wrong password causes no connection. That looks ok.
So you are right about error should be returned when password is given while AuthMethod::None.