esp-rs/esp-wifi-sys #453: Critical section feature conflict between esp-wifi and esp-idf-hal

vortex314, 2024-04-10:
I got the following compiler error 
```
   Compiling critical-section v1.1.2
error: You must set at most one of these Cargo features: restore-state-none, restore-state-bool, restore-state-u8, restore-state-u16, restore-state-u32, restore-state-u64
  --> /home/lieven/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:64:1
   |
64 | compile_error!("You must set at most one of these Cargo features: restore-state-none, restore-state-bool, restore-state-u8, restore-state-u16, restore-state-u32, restore-state-u64");
 
```
I see that esp-wifi wants the feature flag "restore-state-u32" while esp-idf-hal wants the feature flag ""restore-state-none" for crate critical-section.

With the following Cargo.toml 
```toml
[package]
name = "espidf-rust"
version = "0.1.0"
authors = ["lieven <lieven.merckx@gmail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.71"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
default = ["std", "embassy", "esp-idf-svc/native"]

pio = ["esp-idf-svc/pio"]
std = ["alloc", "esp-idf-svc/binstart", "esp-idf-svc/std"]
alloc = ["esp-idf-svc/alloc"]
nightly = ["esp-idf-svc/nightly"]
experimental = ["esp-idf-svc/experimental"]
embassy = ["esp-idf-svc/embassy-sync", "esp-idf-svc/critical-section", "esp-idf-svc/embassy-time-driver"]

[dependencies]
log = { version = "0.4", default-features = false }
esp-idf-svc = { version = "0.48", default-features = false , features = [] }
esp-idf-hal = { version = "0.43.1", default-features = false , features = [] }
minicbor = { version = "0.21.1", features=["derive","std"] }
cobs={ version = "0.2.3", default_features=false,features = [] }
crc = { version = "3.0.1", features = [] }
esp-wifi = { version = "0.4.0" , features = ["esp32","wifi"]}
esp-hal = { version = "0.16.1" , default_features=false,features = []}
anyhow       = "1.0.81"
embedded-svc = "0.27.1"
shtcx        = "=0.11.0"
toml-cfg     = "0.2.0"
critical-section =  { version = "1.1.2", default-features = false}

[build-dependencies]
embuild  = "=0.31.4"
toml-cfg = "0.2.0"
```
With command 
```sh
$ cargo tree -e features
```
I see that esp-wifi wants the feature flag "restore-state-u32" while esp-idf-hal wants the feature flag ""restore-state-none" for crate critical-section.

How to address this ?

bjoernQ, 2024-04-12:
Seems you are trying to use esp-wifi on esp-idf/Rust std. That's not supported (it's for bare-metal Rust). You probably want to use https://crates.io/crates/esp-idf-svc which offers the same functionality (and more) for Rust-Std on esp-idf

MabezDev, 2024-04-12:
As @bjoernQ said, you're trying to mix our two approaches, please take a read of https://docs.esp-rs.org/book/ to familiarize yourself with the approaches we offer - after which you can open a new issue if you run into issues :)

closed 2024-04-12 by MabezDev

vortex314, 2024-04-12:
Ok, clear. Indeed I mangled the 2 different approaches : no_std and std.  Compilation works fine now.
Thanks !