esp-rs/esp-wifi-sys #455: #[interrupt] macro has been removed from esp-hal-procmacros

bbustin, 2024-04-13:
The #[interrupt] macro has been removed from esp-hal-procmacros as of Apr 9, 2024.

Please see the associated pull request for more information:
- https://github.com/esp-rs/esp-hal/pull/1409

I noticed this causes problems in esp-wifi/src/timer/riscv.rs. I have not done a thorough deep-dive; however, I suspect it also would affect xtensa.rs.

I am having some other issues compiling against the latest esp-hal, so will be unable to submit a pull request at this time. If I figure things out, then I would be glad to contribute my findings as a pull request.

bjoernQ, 2024-04-15:
Hello and thanks! Yes, there are some incompatible changes in esp-hal's `main` branch which I'll address soon

If you don't really need the very latest commits from `esp-hal` you should be fine by using `0.16.1` released on crates.io and `esp-wifi` 0.4.0 (also from crates.io)

Otherwise, the needed changes be made here, soon

Firstyear, 2024-04-15:
Somehow this error has leaked into crates.io versions, so I can't build either. :(

bjoernQ, 2024-04-15:
> Somehow this error has leaked into crates.io versions, so I can't build either. :(

Are you sure you don't reference / patched any GIT dependencies?

Firstyear, 2024-04-15:
```
 # cat Cargo.toml
[package]
name = "no-std-template"
version = "0.1.0"
authors = ["William Brown <william@blackhats.net.au>"]
edition = "2021"
license = "MIT OR Apache-2.0"

[profile.dev]
# Rust debug is too slow.
# For debug builds always builds with some optimization
opt-level = "s"

[profile.release]
codegen-units = 1 # LLVM can perform better optimizations using a single thread
debug = 2
debug-assertions = false
incremental = false
lto = 'fat'
opt-level = 's'
overflow-checks = false

[dependencies]
defmt = { version = "0.3.6", optional = true }
esp-hal = { version = "0.16.1", default-features = false }
smoltcp = { version = "0.11.0", default-features = false, features = [
    "medium-ethernet",
    "socket-raw",
], optional = true }
critical-section = "1.1.1"
log = { version = "0.4.20", optional = true }
embedded-hal = { version = "0.2.4", default-features = false }
embedded-svc = { version = "0.27.0", default-features = false, features = [], optional = true }
enumset = { version = "1.1.3", default-features = false, optional = true }
linked_list_allocator = { version = "0.10.5", default-features = false, features = [
    "const_mut_refs",
] }
embedded-io = { version = "0.6.1", default-features = false }
embedded-io-async = { version = "0.6.0", optional = true }
fugit = "0.3.7"
heapless = { version = "0.8", default-features = false, features = [
    "portable-atomic",
] }
num-derive = { version = "0.4" }
num-traits = { version = "0.2", default-features = false }

no-std-net = { version = "0.6.0", optional = true }

esp-wifi = { version = "0.4.0" }
esp-wifi-sys = { version = "0.3.0" }

embassy-sync = { version = "0.5.0", optional = true }
embassy-futures = { version = "0.1.0", optional = true }
embassy-net-driver = { version = "0.2", optional = true }
toml-cfg = "0.2.0"
libm = "0.2.7"
cfg-if = "1.0.0"
portable-atomic = { version = "1.5", default-features = false }
portable_atomic_enum = { version = "0.3.0", features = ["portable-atomic"] }

futures-util = { version = "0.3.28", default-features = false, features = [
    "portable-atomic",
] }
atomic-waker = { version = "1.1.2", default-features = false, features = [
    "portable-atomic",
] }

# Former dev deps

esp-println = { version = "0.9.0", default-features = false, features = [
    "log",
    "uart"
] }
esp-backtrace = { version = "0.11.0" , features = [
    "panic-handler",
    "exception-handler",
    "println",
] }
embassy-executor = { version = "0.5.0", package = "embassy-executor", features = [
    "nightly",
    "integrated-timers",
] }
embassy-time = { version = "0.3.0" }
embassy-net = { version = "0.4.0", features = [
    "tcp",
    "udp",
    "dhcpv4",
    "medium-ethernet",
] }
bleps = { git = "https://github.com/bjoernQ/bleps", package = "bleps", rev = "9371d7d4d510ba5c936c1eef96674f8fd4f63e8a", features = [
    "macros", "async"
] }
embedded-hal-async = { version = "1.0.0" }
static_cell = { version = "2.0", features = ["nightly"] }

[build-dependencies]
toml-cfg = "0.2.0"

[features]
default = [ "log" ]

# chip features
esp32c2 = [ "esp-hal/esp32c2", "esp-wifi-sys/esp32c2", "esp-println/esp32c2", "esp-backtrace/esp32c2" ]
esp32c3 = [ "esp-hal/esp32c3", "esp-wifi-sys/esp32c3", "esp-println/esp32c3", "esp-backtrace/esp32c3" ]
esp32c6 = [ "esp-hal/esp32c6", "esp-wifi-sys/esp32c6", "esp-println/esp32c6", "esp-backtrace/esp32c6" ]
esp32h2 = [ "esp-hal/esp32h2", "esp-wifi-sys/esp32h2", "esp-println/esp32h2", "esp-backtrace/esp32h2" ]
esp32   = [ "esp-hal/esp32",   "esp-wifi-sys/esp32",   "esp-println/esp32",   "esp-backtrace/esp32" ]
esp32s2 = [ "esp-hal/esp32s2", "esp-wifi-sys/esp32s2", "esp-println/esp32s2", "esp-backtrace/esp32s2" ]
esp32s3 = [ "esp-hal/esp32s3", "esp-wifi-sys/esp32s3", "esp-println/esp32s3", "esp-backtrace/esp32s3" ]

# async features
async = [
  "dep:embassy-sync",
  "dep:embassy-futures",
  "dep:embedded-io-async",
  "esp-hal/embassy",
  "esp-hal/async",
]

embassy-net = ["dep:embassy-net-driver", "async"]

# misc features
coex = []
wifi-logs = []
dump-packets = []
smoltcp = [ "dep:smoltcp" ]
utils = [ "smoltcp" ]
enumset = []
wifi = [ "dep:enumset", "dep:no-std-net" ]
embedded-svc = [ "dep:embedded-svc" ]
ble = [ "esp-hal/bluetooth" ]
phy-enable-usb = []
ps-min-modem = []
ps-max-modem = []
esp-now = [ "wifi" ]
ipv6   = ["wifi", "utils", "smoltcp?/proto-ipv6"]
ipv4   = ["wifi", "utils", "smoltcp?/proto-ipv4"]
tcp    = ["ipv4", "smoltcp?/socket-tcp"]
udp    = ["ipv4", "smoltcp?/socket-udp"]
icmp   = ["ipv4", "smoltcp?/socket-icmp"]
igmp   = ["ipv4", "smoltcp?/proto-igmp"]
dns    = ["udp",  "smoltcp?/proto-dns", "smoltcp?/socket-dns"]
dhcpv4 = ["wifi", "utils", "smoltcp?/proto-dhcpv4", "smoltcp?/socket-dhcpv4"]
wifi-default = ["ipv4", "tcp", "udp", "icmp", "igmp", "dns", "dhcpv4"]
defmt = [
  "dep:defmt",
  "smoltcp?/defmt",
  "esp-hal/defmt",
]
log = [
  "dep:log",
  "esp-hal/log",
]

[package.metadata.docs.rs]
features = ["esp32c3", "wifi", "ble", "coex", "async", "embassy-net", "esp-hal/embassy-time-systick", "esp-hal/default"]
default-target = "riscv32imc-unknown-none-elf"
```

I have been trying for a few days to get the esp-wifi examples to work, and I've been encountering issue after issue :( 

This Cargo.toml was built from a combination of the no-std-template from cargo-generate and by merging the content of the esp-wifi workspace and crate Cargo.toml.

bjoernQ, 2024-04-15:
I see it's currently not easy to get dependencies and features right.

I prepared a standalone example using esp-wifi (async) here: https://github.com/bjoernQ/esp-hal-esp-wifi-example

It's configured for ESP32-C3 and basically the `embassy_dhcp.rs` example

Firstyear, 2024-04-15:
Yeah, this is being discussed over at https://github.com/esp-rs/esp-hal/issues/1432#issuecomment-2056876192 

And yeah, it's not just "not easy", I'd say it's "nearly impossible" :( I'd love to see it improve too. Thanks!

PR #456 (2024-04-19): Prepare next esp-hal
Closes #455 

This is in preparation for the next HAL release. Turns out this needed more changes than expected
closed 2024-04-19 by bjoernQ