esp-rs/esp-wifi-sys #461: docs.rs failed to build

phemmer, 2024-04-20:
Docs on docs.rs are currently failing to build: https://docs.rs/crate/esp-wifi/0.5.0/builds/1194959/riscv32imc-unknown-none-elf.txt

```
[INFO] [stderr] error: failed to select a version for `esp-hal`.
[INFO] [stderr]     ... required by package `esp-wifi v0.5.0 (/opt/rustwide/workdir)`
[INFO] [stderr] versions that meet the requirements `^0.17.0` (locked to 0.17.0) are: 0.17.0
[INFO] [stderr] 
[INFO] [stderr] the package `esp-wifi` depends on `esp-hal`, with features: `embassy-time-systick` but `esp-hal` does not have these features.
```

Looks like issue is here: https://github.com/esp-rs/esp-wifi/blob/a62d7f41426fee073bfaf045f2b1d647397078c0/esp-wifi/Cargo.toml#L129 as [`embassy-time-systick` was removed](https://github.com/esp-rs/esp-hal/commit/e65951c?diff=unified&w=0#diff-2c38757b7440c0eee9b3ad946e948db36e91f5d1fb10e60daaddf81bc2ca162bL183-R190)

MabezDev, 2024-04-20:
@bjoernQ I think we need a patch release from main, as you've already updated the features for doc.rs in main (we should probably add a CI check for esp-wifi and esp-hal using cargo docs-rs).

@phemmer Sorry about this, we'll get the docs working next week, for now you can build the docs locally from main.

bjoernQ, 2024-04-22:
I was thinking if it's worth a patch release but probably it is

MabezDev, 2024-04-22:
> I was thinking if it's worth a patch release but probably it is

Yeah until we self host the docs like esp-hal, having the docs available is a must. Thanks for taking care of it!

closed 2024-04-22 by MabezDev