esp-rs/esp-wifi-sys #468: 802.11r support?

sivakov512, 2024-05-11:
Hi! After few hours I found that my esp32c3 can't connect to wifi ap if 802.11r enabled. Looks like it's possible to enable ([doc](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/wifi.html#fast-bss-transition)), but please consult me how? Can't find any info about it

bjoernQ, 2024-05-14:
One thing you could try is to change `apply_sta_config` in `src\wifi\mod.rs`  to set `ft_enable`

You could just make the unsafe block look like this:
```rust
    unsafe {
        cfg.sta.ssid[0..(config.ssid.len())].copy_from_slice(config.ssid.as_bytes());
        cfg.sta.password[0..(config.password.len())].copy_from_slice(config.password.as_bytes());

        cfg.sta.set_ft_enabled(1); // added

        esp_wifi_result!(esp_wifi_set_config(wifi_interface_t_WIFI_IF_STA, &mut cfg))
    }
```

Probably it will not be enough and we would need to change how the supplicant is compiled. Unfortunately, I don't have access to an 802.11r capable AP to assist more with this

sivakov512, 2024-05-14:
> Probably it will not be enough and we would need to change how the supplicant is compiled. 

is it hard? If not so then can you explain? I can try and test it within my network

bjoernQ, 2024-05-14:
Not too hard I guess

There is https://github.com/esp-rs/esp-wireless-drivers-3rdparty - you should use esp-idf of the same version ideally ( https://github.com/esp-rs/esp-wireless-drivers-3rdparty/blob/master/version )

In the sdkconfig.defaults (per chip, e.g. https://github.com/esp-rs/esp-wireless-drivers-3rdparty/blob/master/patch/esp32c3/sdkconfig.defaults ) you need to add `CONFIG_ESP_WIFI_11R_SUPPORT=y` and run the makefile (with the esp-idf installation activated)

In the end you should have a different `libwpasupplicant.a` for each chip - everything else should be the same. Just copy over the supplicant to esp-wifi-sys

sivakov512, 2024-05-14:
Ok, I will try. After that I can just make PR with new blobs, right?
Is changes you show earlier also required?

bjoernQ, 2024-05-14:
> Ok, I will try. After that I can just make PR with new blobs, right? Is changes you show earlier also required?

Yes, the changes to the code are also required but we should have that configurable somehow. Ideally via https://docs.rs/esp-wifi/0.5.1/esp_wifi/wifi/struct.ClientConfiguration.html