esp-rs/esp-wifi-sys #470: BLE and Wifi examples not working on ESP32S3.

hydra, 2024-05-13:
Hi,

I'm testing the examples on board here, and none of the wifi or bluetooth examples that I tried work.

I was initially using the cargo-generate code and after initialising the wifi the code just crashed, so when that didn't work I tried the examples here to rule out any of my code and got an identical issue.

Here's logs:

```
$ cargo esp32s3 --release --example access_point --features=wifi -- --before no-reset
    Finished release [optimized + debuginfo] target(s) in 0.66s
     Running `espflash flash --monitor D:\Users\Hydra\Documents\dev\libraries\rust\esp-wifi\target\xtensa-esp32s3-none-elf\release\examples\access_point --before no-reset`
[2024-05-13T22:27:42Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-05-13T22:27:42Z INFO ] Detected 4 serial ports
[2024-05-13T22:27:42Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-13T22:27:42Z INFO ] Please select a port
[2024-05-13T22:27:43Z INFO ] Serial port: 'COM15'
[2024-05-13T22:27:43Z INFO ] Connecting...
[2024-05-13T22:27:44Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:54:75:d8:7e:e4
App/part. size:    466,912/1,048,576 bytes, 44.53%
[2024-05-13T22:27:46Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-05-13T22:27:46Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=00029584 vaddr=00000000 size=06a94h ( 27284)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeou
```

for this log, note that the output is truncated, the last log message is incomplete, no further output occurs.

[... 161 lines cut ...]
        esp_wifi::EspWifiInitFor::Wifi,
        timer,
        esp_hal::rng::Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();
    let wifi = peripherals.WIFI;

    log::info!("C");
    let (iface, device, mut controller, sockets) =
        create_network_interface(&init, wifi, WifiApDevice, socket_set_entries_static_ref).unwrap();

```

log from my code:

```
INFO - A
INFO - B
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
```
and nothing else is output, and no AP appears (note that "INFO - C" is never printed.

Any suggestions?

hydra, 2024-05-13:
I was able to run the 'access_point' example on a ESP32-WROOM-32 module OK, as follows:

```
$ cargo esp32 --release --example access_point --features=wifi
    Finished release [optimized + debuginfo] target(s) in 0.67s
     Running `espflash flash --monitor D:\Users\Hydra\Documents\dev\libraries\rust\esp-wifi\target\xtensa-esp32-none-elf\release\examples\access_point`
[2024-05-13T22:57:01Z INFO ] Detected 5 serial ports
[2024-05-13T22:57:01Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-13T22:57:01Z INFO ] Please select a port
[2024-05-13T22:57:03Z INFO ] Serial port: 'COM3'
[2024-05-13T22:57:03Z INFO ] Connecting...
[2024-05-13T22:57:04Z INFO ] Using flash stub
Chip type:         esp32 (revision v1.0)
Crystal frequency: 40 MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       3c:71:bf:63:cf:d8
App/part. size:    497,616/4,128,768 bytes, 12.05%
[2024-05-13T22:57:05Z INFO ] Segment at address '0x1000' has not changed, skipping write
[2024-05-13T22:57:05Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-05-13T22:57:05Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-05-13T22:57:06Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7104
0x3fff0030 - esp_wifi::HEAP_DATA
    at ??:??
load:0x40078000,len:15576
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
0x4008064c - <&mut W as core::fmt::Write::write_fmt::SpecWriteFmt>::spec_write_fmt
    at D:\Users\Hydra\.rustup\toolchains\esp\lib\rustlib\src\rust\library\core\src\fmt\mod.rs:211
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 07:48:23
I (32) boot: Multicore bootloader
I (37) boot: chip revision: v1.0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 003f0000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=12c90h ( 76944) map
I (133) esp_image: segment 1: paddr=00022cb8 vaddr=3ffb0000 size=00e10h (  3600) load
I (135) esp_image: segment 2: paddr=00023ad0 vaddr=3ffb9314 size=00190h (   400) load
I (139) esp_image: segment 3: paddr=00023c68 vaddr=40080000 size=0c3b0h ( 50096) load
I (168) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=58810h (362512) map
I (300) esp_image: segment 5: paddr=00088838 vaddr=4008c3b0 size=00f74h (  3956) load
I (308) boot: Loaded app from partition at offset 0x10000
I (308) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
wifi_set_configuration returned Ok(())
Error:   × The device does not recognize the command. (os error 22)

error: process didn't exit successfully: `espflash flash --monitor D:\Users\Hydra\Documents\dev\libraries\rust\esp-wifi\target\xtensa-esp32-none-elf\release\examples\access_point` (exit code: 1)
```
[... 84 lines cut ...]
```
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
```

vs ESP32-S3-WROOM-1:
```
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.1
```

For reference, the markings on the S3 on my board are as follows:
```
ESP32-S3-PICO-1
202023
N8RC2018154
MNP632000
FC00EAE007
```

Can anyone else confirm that `esp-wifi` works for them on an ESP32-S3 with chip revision 0.2.

JurajSadel, 2024-05-14:
Hi @hydra, I tested examples you've mentioned, and I can confirm they work fine for me (tested with esp32s3 revision 0.2).

JurajSadel, 2024-05-14:
@hydra Sure, I'm using the `ESP32-S3-DevkitC-1 v1.1`

Logs:

```rust
cargo esp32s3 --release --example access_point --features=wifi
    Finished release [optimized + debuginfo] target(s) in 0.10s
     Running `espflash flash --monitor /Users/jurajsadel/esp-rs/esp-wifi/target/xtensa-esp32s3-none-elf/release/examples/access_point`
[2024-05-14T14:32:04Z INFO ] Detected 2 serial ports
[2024-05-14T14:32:04Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-14T14:32:04Z INFO ] Please select a port
[2024-05-14T14:32:05Z INFO ] Serial port: '/dev/cu.usbserial-1140'
[2024-05-14T14:32:05Z INFO ] Connecting...
[2024-05-14T14:32:05Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:da:0c:61:dd:b0
App/part. size:    466,912/1,048,576 bytes, 44.53%
[2024-05-14T14:32:07Z INFO ] Segment at address '0x0' has not changed, skipping write
[2024-05-14T14:32:07Z INFO ] Segment at address '0x8000' has not changed, skipping write
[2024-05-14T14:32:08Z INFO ] Segment at address '0x10000' has not changed, skipping write
[2024-05-14T14:32:08Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=12224h ( 74276) map
I (123) esp_image: segment 1: paddr=0002224c vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=0002327c vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=000233ec vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=0002962c vaddr=00000000 size=069ech ( 27116)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Ok(EnumSet(AccessPoint))
INFO - Set hardware address: Ethernet(Address([222, 218, 12, 97, 221, 176]))
Start busy loop on main. Connect to the AP `esp-wifi` and point your browser to http://192.168.2.1:8080/
Use a static IP in the range 192.168.2.2 .. 192.168.2.255, use gateway 192.168.2.1
Connected
GET / HTTP/1.1
Host: 192.168.2.1:8080
Upgrade-Insecure-Requests: 1
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/17.4.1 Safari/605.1.15
Accept-Language: en-US,en;q=0.9
Accept-Encoding: gzip, deflate
Connection: keep-alive

Done

```
Sorry for closing the issue, I thought it could eb closed after verifying that examples are working correctly.

hydra, 2024-05-14:
@JurajSadel thanks for the log, i'll diff it to mine and see if there's anything obviously wrong, I also have 2 other boards with S3's on and will test them and report back.

hydra, 2024-05-14:
So another board I have here, with a chip from the same batch of MCUs runs the example just fine...

```
$ cargo esp32s3 --release --example access_point --features=wifi -- --before no-reset
    Finished release [optimized + debuginfo] target(s) in 0.67s
     Running `espflash flash --monitor D:\Users\Hydra\Documents\dev\libraries\rust\esp-wifi\target\xtensa-esp32s3-none-elf\release\examples\access_point --before no-reset`
[2024-05-14T16:40:18Z WARN ] Pre-connection option 'NoReset' was selected. Connection may fail if the chip is not in bootloader or flasher stub mode.
[2024-05-14T16:40:18Z INFO ] Detected 4 serial ports
[2024-05-14T16:40:18Z INFO ] Ports which match a known common dev board are highlighted
[2024-05-14T16:40:18Z INFO ] Please select a port
[2024-05-14T16:40:23Z INFO ] Serial port: 'COM15'
[2024-05-14T16:40:23Z INFO ] Connecting...
[2024-05-14T16:40:24Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       dc:54:75:d8:7e:f8
App/part. size:    466,912/1,048,576 bytes, 44.53%
[00:00:01] [========================================]      14/14      0x0
[00:00:00] [========================================]       1/1       0x8000
[00:00:28] [========================================]     286/286     0x10000
[2024-05-14T16:40:56Z INFO ] Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=00029584 vaddr=00000000 size=06a94h ( 27284)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Ok(EnumSet(AccessPoint))
INFO - Set hardware address: Ethernet(Address([222, 84, 117, 216, 126, 248]))
Start busy loop on main. Connect to the AP `esp-wifi` and point your browser to http://192.168.2.1:8080/
Use a static IP in the range 192.168.2.2 .. 192.168.2.255, use gateway 192.168.2.1
```

most odd indeed!

I'll continue to identify the differences and will also try a few more boards here in case there's something wrong with the first one I tried.

bjoernQ, 2024-05-15:
One thing which might be worth checking would be the power-supply. Curious about the outcome of this

hydra, 2024-05-15:
A the 2nd board from the same panel as the first one works just fine, indicating that the power supply circuit must be ok.  I use the same components, regulators, caps, etc, on other ESP32S3 and D4 designs so I know they're good.  I'll do more testing and see if I can identify the issue with the first board and report back with findings.

This is the panel, the one that I was having issues with is the one in the lower left.

[image]

hydra, 2024-05-15:
The issue seems to be intermittent.

```

    let timer = esp_hal::timer::TimerGroup::new(peripherals.TIMG1, &clocks, None).timer0;
    let init = esp_wifi::initialize(
        esp_wifi::EspWifiInitFor::Wifi,
        timer,
        esp_hal::rng::Rng::new(peripherals.RNG),
        system.radio_clock_control,
        &clocks,
    )
    .unwrap();
    let wifi = peripherals.WIFI;

    wait_for_button(&mut rwdt);

    let (iface, device, mut controller, sockets) =
        create_network_interface(&init, wifi, WifiApDevice, socket_set_entries_static_ref).unwrap();

    wait_for_button(&mut rwdt);

    let mut wifi_stack = WifiStack::new(iface, device, sockets, current_millis);

    wait_for_button(&mut rwdt);

    let client_config = Configuration::AccessPoint(AccessPointConfiguration {
        ssid: ssid.as_str().try_into().unwrap(),
        ..Default::default()
    });
    let res = controller.set_configuration(&client_config);
    log::info!("wifi_set_configuration returned {:?}", res);

    wait_for_button(&mut rwdt);

    controller.start().unwrap();
    // <-- control not returned here
    log::info!("is wifi started: {:?}", controller.is_started());

    wait_for_button(&mut rwdt);

    log::info!("{:?}", controller.get_capabilities());

    wait_for_button(&mut rwdt);
```

yields:
```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
INFO - Waiting for button press
INFO - !B!
INFO - Waiting for button press
INFO - !B!
INFO - Waiting for button press
INFO - !B!
INFO - wifi_set_configuration returned Ok(())
INFO - Waiting for button press
INFO - !B!
```

Corresponding scope trace:
[image]

Yellow = 3v3, Cyan = 5V, Blue = IO0 (button)

hydra, 2024-05-15:
When it works, it looks like this - scope zoomed in a bit to show detail.

log:
```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
INFO - Waiting for button press
INFO - !B!
INFO - Waiting for button press
INFO - !B!
INFO - Waiting for button press
INFO - !B!
INFO - wifi_set_configuration returned Ok(())
INFO - Waiting for button press
INFO - !B!
INFO - is wifi started: Ok(true)
INFO - Waiting for button press
INFO - !B!
INFO - Ok(EnumSet(AccessPoint))
INFO - Waiting for button press
INFO - !B!
INFO - Set hardware address: Ethernet(Address([222, 84, 117, 216, 126, 184]))
```

[image]

hydra, 2024-05-15:
And here's a log showing that it's intermittent:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=00029584 vaddr=00000000 size=06a94h ( 27284)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
wifi_set_configuration returned Ok(())
is wifi started: Ok(true)
Ok(EnumSet(AccessPoint))
INFO - Set hardware address: Ethernet(Address([222, 84, 117, 216, 126, 184]))
Start busy loop on main. Connect to the AP `esp-wifi` and point your browser to http://192.168.2.1:8080/
Use a static IP in the range 192.168.2.2 .. 192.168.2.255, use gateway 192.168.2.1
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x29 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3818,len:0x16f8
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (30) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (30) boot: compile time Jun  7 2023 08:07:32
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
[... 24 lines cut ...]
I (35) boot: chip revision: v0.2
I (39) boot.esp32s3: Boot SPI Speed : 40MHz
I (44) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c060020 size=1217ch ( 74108) map
I (123) esp_image: segment 1: paddr=000221a4 vaddr=3fc8e238 size=01028h (  4136) load
I (125) esp_image: segment 2: paddr=000231d4 vaddr=3fca7670 size=00168h (   360) load
I (129) esp_image: segment 3: paddr=00023344 vaddr=40378000 size=06238h ( 25144) load
I (145) esp_image: segment 4: paddr=00029584 vaddr=00000000 size=06a94h ( 27284)
I (152) esp_image: segment 5: paddr=00030020 vaddr=42000020 size=51f90h (335760) map
I (240) boot: Loaded app from partition at offset 0x10000
I (240) boot: Disabling RNG early entropy source...
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout:
```

[image]

On the scope you can see the wifi initialized on the first 2 attempts, and the logs show it too, but on the 3rd attempt it dies in `controller.start().unwrap();` before the MCU has had the chance to clock-out the remaining UART buffer.

hydra, 2024-05-15:
I also tried adding an extra 10uF capacitor to the 3v3 reg's output, made no different.
I tried adding hacking i n`esp_wifi_result!(esp_wifi_sys::include::esp_wifi_set_max_tx_power(8))?;` (`8` is the lowest power setting) into `wifi_start()` which makes a noticable difference on the 3v3 rail but it's still intermittent.

tomorrow I'll try hooking up a JTAG debugger to see if that sheds more light on things.

I could also try just populating another board without anything connected to the S3 other than the regs, caps and other required components, but that's a bit of a faff and involves a full day of manufacturing and mucking around with my PnP machine...

If anyone has any other thoughts of things to try I'm all ears.

hydra, 2024-05-16:
Seems like when it gets stuck, it gets stuck like this:

```
(gdb) info frame
Stack level 0, frame at 0x3fc91ec0:
 pc = 0x4201b7c4 in esp_hal::critical_section_impl::multicore::ReentrantMutex::lock (src/lib.rs:420); saved pc = 0x4200f543
 inlined into frame 1
 source language rust.
 Arglist at unknown address.
 Locals at unknown address, Previous frame's sp in a1
(gdb) info stack
#0  esp_hal::critical_section_impl::multicore::ReentrantMutex::lock (self=0x3fc8e3b0 <esp_hal::critical_section_impl::multicore::MULTICORE_LOCK>) at src/lib.rs:420
#1  esp_hal::critical_section_impl::xtensa::{impl#0}::acquire () at src/lib.rs:296
#2  0x4200f543 in critical_section::acquire () at D:\Users\Hydra\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src/lib.rs:183
#3  critical_section::with<core::option::Option<[u8; 8]>, esp_wifi::compat::common::receive_queued::{closure_env#0}> () at D:\Users\Hydra\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src/lib.rs:226
#4  esp_wifi::compat::common::receive_queued (queue=<optimized out>, item=<optimized out>, block_time_tick=4294967295) at esp-wifi\src\compat/common.rs:293
#5  0x4200d7ff in esp_wifi::wifi::os_adapter::queue_recv (queue=0x3fc942e4 <_ZN8esp_wifi5timer13arch_specific15TIMER_OVERFLOWS17h7090171393f353e8E.llvm.8773300724186734139>, item=0x3fca45fc <log::MAX_LOG_LEVEL_FILTER>, block_time_tick=1107408804) at esp-wifi\src\wifi/os_adapter.rs:515
#6  0x4037a648 in ppTask ()
#7  0x42010734 in esp_wifi::compat::task_runner::run_c_task () at esp-wifi\src\compat/task_runner.rs:47
(gdb) info threads
  Id   Target Id         Frame 
* 1    Remote target     esp_hal::critical_section_impl::multicore::ReentrantMutex::lock (self=0x3fc8e3b0 <esp_hal::critical_section_impl::multicore::MULTICORE_LOCK>) at src/lib.rs:420
(gdb) info registers
pc             0x4201b7c4          0x4201b7c4 <esp_hal::critical_section_impl::xtensa::{impl#0}::acquire+32>
lbeg           0x40056f5c          1074098012
lend           0x40056f72          1074098034
lcount         0x0                 0
sar            0x3                 3
ps             0x60105             393477
threadptr      0x0                 0
br             0x0                 0
scompare1      0x100               256
acclo          0x0                 0
acchi          0x0                 0
m0             0x0                 0
m1             0x0                 0
m2             0x0                 0
m3             0x0                 0
fcr            0x0                 0
fsr            0x0                 0
pwrctl         0x0                 0
pwrstat        0x0                 0
eristat        0x0                 0
cs_itctrl      0x0                 0
cs_claimset    0x0                 0
cs_claimclr    0x0                 0
cs_lockaccess  0x0                 0
cs_lockstatus  0x0                 0
cs_authstatus  0x0                 0
fault_info     0x0                 0
trax_id        0x0                 0
trax_control   0x0                 0
trax_status    0x0                 0
trax_data      0x0                 0
trax_address   0x0                 0
trax_pctrigger 0x0                 0
trax_pcmatch   0x0                 0
trax_delay     0x0                 0
trax_memstart  0x0                 0
trax_memend    0x0                 0
pmg            0x0                 0
pmpc           0x0                 0
pm0            0x0                 0
pm1            0x0                 0
pmctrl0        0x0                 0
pmctrl1        0x0                 0
pmstat0        0x0                 0
pmstat1        0x0                 0
ocdid          0x0                 0
ocd_dcrclr     0x0                 0
ocd_dcrset     0x0                 0
ocd_dsr        0x0                 0
a0             0x8200f543          -2113866429
a1             0x3fc91ea0          1070145184
a2             0x60105             393477
a3             0xffffffff          -1
a4             0x3fc9001c          1070137372
a5             0x3c0678ac          1007057068
a6             0x3c0678ac          1007057068
a7             0x0                 0
a8             0x0                 0
a9             0x100               256
a10            0x3fc8e3b0          1070130096
a11            0x0                 0
a12            0x803793f5          -2143841291
a13            0x3fcda910          1070442768
a14            0x2                 2
a15            0x35                53
```

[image]

in `esp_wifi::compat::common::receive_queued`

I'll try and turn on some more logging.

hydra, 2024-05-16:
here's what it looks like with `wifi-logs' enabled, when it fails:
```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
INFO - pp rom version: e7ae62f

INFO - net80211 rom version: e7ae62f

INFO - I (2231) wifi:
INFO - wifi driver task: 00, prio:253, stack:6656, core=0
INFO -

INFO - I (2239) wifi:
INFO - wifi firmware version: e309346
INFO -

INFO - I (2244) wifi:
INFO - wifi certification version: v7.0
INFO -

INFO - I (2250) wifi:
INFO - config NVS flash: disabled
INFO -

INFO - I (2256) wifi:
INFO - config nano formating: disabled
INFO -

INFO - I (2262) wifi:
INFO - Init data frame dynamic rx buffer num: 32
INFO -

INFO - I (2269) wifi:
INFO - Init static rx mgmt buffer num: 10
INFO -

INFO - I (2275) wifi:
INFO - Init management short buffer num: 32
INFO -

INFO - I (2282) wifi:
INFO - Init dynamic tx buffer num: 32
INFO -

INFO - I (2288) wifi:
INFO - Init static tx FG buffer num: 2
INFO -

INFO - I (2294) wifi:
INFO - Init static rx buffer size: 2212
INFO -

INFO - I (2300) wifi:
INFO - Init static rx buffer num: 10
INFO -

INFO - I (2306) wifi:
INFO - Init dynamic rx buffer num: 32
INFO -

INFO - V (2312) wifi:
INFO - hint=255, act_dur=<0,120>, pas_dur=360, nchans=14 history_num=0
INFO -

INFO - W (2322) wifi:
INFO - Affected by the ESP-NOW encrypt num, set the max connection num to 10
INFO -

INFO - W (2331) wifi:
INFO - Affected
```

and when it works:
```
INFO - esp-wifi configuration Config { rx_queue_size: 5, tx_queue_size: 3, static_rx_buf_num: 10, dynamic_rx_buf_num: 32, static_tx_buf_num: 0, dynamic_tx_buf_num: 32, ampdu_rx_enable: 0, ampdu_tx_enable: 0, amsdu_tx_enable: 0, rx_ba_win: 6, max_burst_size: 1, country_code: "CN", country_code_operating_class: 0, mtu: 1492, heap_size: 65536, tick_rate_hz: 100, listen_interval: 3, beacon_timeout: 6, ap_beacon_timeout: 300, failure_retry_cnt: 1, scan_method: 0 }
INFO - pp rom version: e7ae62f

INFO - net80211 rom version: e7ae62f

INFO - I (2232) wifi:
INFO - wifi driver task: 00, prio:253, stack:6656, core=0
INFO -

INFO - I (2241) wifi:
INFO - wifi firmware version: e309346
INFO -

INFO - I (2246) wifi:
INFO - wifi certification version: v7.0
INFO -

INFO - I (2252) wifi:
INFO - config NVS flash: disabled
INFO -

INFO - I (2257) wifi:
INFO - config nano formating: disabled
INFO -

INFO - I (2264) wifi:
INFO - Init data frame dynamic rx buffer num: 32
INFO -

INFO - I (2270) wifi:
INFO - Init static rx mgmt buffer num: 10
INFO -

INFO - I (2277) wifi:
INFO - Init management short buffer num: 32
INFO -

INFO - I (2283) wifi:
INFO - Init dynamic tx buffer num: 32
INFO -

INFO - I (2289) wifi:
INFO - Init static tx FG buffer num: 2
INFO -

INFO - I (2295) wifi:
INFO - Init static rx buffer size: 2212
INFO -

INFO - I (2301) wifi:
INFO - Init static rx buffer num: 10
INFO -

INFO - I (2307) wifi:
INFO - Init dynamic rx buffer num: 32
INFO -

INFO - V (2314) wifi:
INFO - hint=255, act_dur=<0,120>, pas_dur=360, nchans=14 history_num=0
INFO -

INFO - W (2323) wifi:
INFO - Affected by the ESP-NOW encrypt num, set the max connection num to 10
INFO -

INFO - W (2332) wifi:
INFO - Affected by the ESP-NOW encrypt num, set the max connection num to 10
INFO -

wifi_set_configuration returned Ok(())
INFO - V (2371) wifi:
INFO - ht40u freq=2422, chan=1
INFO -

INFO - D (2372) wifi:
INFO - filter: set rx policy=0
INFO -

INFO - I (2373) wifi:
INFO - mode : softAP (de:54:75:d8:7e:b8)
INFO -

INFO - D (2377) wifi:
INFO - filter: set rx policy=9
INFO -

INFO - V (2383) wifi:
INFO - ht40u freq=2422, chan=1
INFO -

INFO - I (2389) wifi:
INFO - Total power save buffer number: 16
INFO -

INFO - I (2394) wifi:
INFO - Init max length of beacon: 752/752
INFO -

INFO - I (2401) wifi:
INFO - Init max length of beacon: 752/752
INFO -

INFO - I (2408) wifi:
INFO - Set ps type: 0, coexist: 0

INFO -

INFO - I (2413) wifi:
INFO - set country: cc=CN schan=1 nchan=13 policy=1

INFO -

is wifi started: Ok(true)
Ok(EnumSet(AccessPoint))
INFO - Set hardware address: Ethernet(Address([222, 84, 117, 216, 126, 184]))
Start busy loop on main. Connect to the AP `esp-wifi` and point your browser to http://192.168.2.1:8080/
Use a static IP in the range 192.168.2.2 .. 192.168.2.255, use gateway 192.168.2.1
```

hydra, 2024-05-16:
It seems that the crash is occuring in the code for `register_chipv7_phy_init_param`, see above stack trace.

For good measure, I erased the entire flash, after reading that the RF calibration data might be stored there, though with regards to the `CONFIG_ESP_PHY_INIT_DATA_IN_PARTITION` I'm not familiar with the interaction between the ROM functions and the rust ESP libs. https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-guides/RF_calibration.html#_CPPv429esp_phy_erase_cal_data_in_nvsv or if it was actually stored there.  I figure it's probably good practice to fully erase the flash to be sure there wasn't some potentially incorrect RF calibration data being used.

Also, after reading that RF calibration can use a lot of power (https://github.com/espressif/esp-idf/issues/7324) I decided to bypass the 5V regulator entirely and supply the S3 MCU with 3.3V directly from the bench PSU.  However, I still get the same intermittent result.

hydra, 2024-05-16:
@JurajSadel Please can you re-test your board and
a) cold-boot it a few times, while monitoring the output of the CP2102 UART port.
b) reboot it a few times, using the reset button, while monitoring the output of the CP2102 UART port.

I'm curious to know if your Rev v0.2 board is also intermittent.

> Sorry for closing the issue, I thought it could eb closed after verifying that examples are working correctly.

please can you re-open this?

hydra, 2024-05-18:
I tested the same exact code on a board with a different design I made which uses the exact same regulators, capcitors, inductors, power supply scheme and so on and everything works just fine, here's the log of me power-cycling it a bunch of times.

[power-cycle-boot-log.txt](https://github.com/esp-rs/esp-wifi/files/15363422/power-cycle-boot-log.txt)

 I also tried:
* adding extra PSU output capactance at the 3v3 reg output (extra 10uF cap stacked in parallel on the existing 10uF cap)
* powering the S3 MCU directly from a bench PSU at the various different VDD pins (VDD3V3, VDDA, VDD_CPU.
* adding extra input capacitance at the VDDA pin (replaced 2.2uF cap with 10uF).

The power traces are wide, using minimum 2 vias per power trace, etc as per the ESP32 S3 hardware guidelines. Using a mostly star topology.

One other issue I have with the boards is that the I2C communication interface is fine at say 4Khz, but over 16Khz results in I2C timeouts and ack failures, I have no idea if this is related.

I'll also try populating a fresh PCB with just the regs, and mcu as I mentioned previously to see if the issue is related to some other components (such as the 3 I2C components)

For good measure, I'll try some other non-esp-rust wifi example software and see how that works.

hydra, 2024-05-18:
Here's a fresh install of the ArduinoIDE and the WifiAP example, it runs just fine.

[image]

Also, with the arduino I2C communication to the M10Q UBlox GPS at 400,000kbit works flawlessly without locking up, which again is probably related, as follows:

[image]

See: https://github.com/esp-rs/esp-hal/pull/1184#issuecomment-2106194991

@JurajSadel please re-open this PR as it looks like there's a software issue either in esp-wifi or with the esp-hal libs to me.  I can send a board to any interested developer that wants to work on this issue.

JurajSadel, 2024-05-23:
Hi @hydra, sorry for late reply. Sure, I can reopen, however, our time is limited and it looks like a weird issue. I can't say right now, when someone will have adequate amount of time to look into this.

hydra, 2024-05-24:
> Hi @hydra, sorry for late reply. Sure, I can reopen, however, our time is limited and it looks like a weird issue. I can't say right now, when someone will have adequate amount of time to look into this.

Hi, thanks for re-opening it, there was probably quite a lot of noise/notifcations while I was digging into it it.

If you have any ideas on where you think I or someone else could start investigating why this might not work please let me know.  I was thinking of comparing the startup code between the arduino code where wifi works and the esp rust crates where it's not working and seeing if I can spot anything obvious/missing/different.